realsenseai/librealsense #1748: How to Actually Use the Examples Given

joshuaguinness, 2018-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.11.0 | 

### Issue Description

New Programmer Here!

I am trying to use the examples given, specifically the C distance example and I don't really have a clue how to use it at all. Do I need to download an IDE? Run it from command line? Do something else? 

Thanks so much for your help!

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
Hi @joshuaguinness,

Please follow below instructions to build the examples:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
Hi @joshuaguinness,

Please use VS2015 or VS2017 to open the project "librealsense2.sln" which is created by CMAKE tool after "Generate". Then, you can build all to generate the sample apps thru VS2015 or VS2017. You can also show me your CMAKE Configuration GUI for librealsense build if you still have the trouble.

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
Hi @joshuaguinness,

Any update? Are you able to run the sample apps?

joshuaguinness, 2018-06-05:
Thanks for checking up. I have been busy with some other things at the moment but I plan to return to this project within the next couple of days and will update then.

joshuaguinness, 2018-06-19:
I am still having trouble running the sample apps. This is what my CMAKE Configuration GUI looks like. 

[image]

This is what it looks like after I clicked Configure. From what I can see, it looks like it both Configured and Generated in one step.

[image]

I then navigated to where it was stored on my computer and selected "librealsense2.sln" and ran it using VS2017.

I then navigated to the Solution Explorer, right clicked on rs-distance, and selected build.

[image]

After building it gave me this error.

[image]

I also tried right clicking on ALL BUILD and then clicking build but got this error.

[image]

If you could point me in the right direction that would be amazing! Thanks.

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
Hi @joshuaguinness,

From the pictures, it looks some files like 'atlcomcli.h' are missing. And, I saw you use VS2017 Community version, not official (professional or enterprise) version. This could be the problem to cause those missing files. Please try to install the official version and check if it works.

RealSense-Customer-Engineering, 2018-07-01:
[Realsense Customer Engineering Team Comment]
Hi @joshuaguinness,

Is your issue resolved? Still need any support for this topic?

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Hi @joshuaguinness,

I will close this first. You can still re-open or create another ticket if any more issues.

closed 2018-07-09 by RealSense-Customer-Engineering