realsenseai/librealsense #1757: Realsense tools build with user level (-DFORCE_LIBUVC=ON), fail when executing.

mjscosta, 2018-05-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-43-generic  | 
| Platform                                 | PC etc..  |
| SDK Version                            |  2.11.1  | 

### Issue Description
Installing dkms modules, implies that the user must have admin permissions on the system. After instaling latest dkms modules into ubuntu 16.04 and having the pc crashed, this raised some questions about distributing applications for realsense cameras (SR300/D400) that will build modules into users kernels and compromise the stability of the same.

So when trying to address the problem using user space drivers , compiling with:
`cmake -DFORCE_LIBUVC=ON -DUSE_SYSTEM_LIBUSB=ON ../`

Executing rs-enumerate-devices  and other tools lead to an error that seems to be related to udev permissions, see log below.

```
./rs-enumerate-devices 
 23/05 17:25:39,704 ERROR [140408896206656] (types.h:179) Could not open device. Last Error: Permission denied
Segmentation fault (core dumped)
```

using gdb with the core dump results in the following log:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./rs-enumerate-devices'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f7790c802a6 in libusb_submit_transfer () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
[Current thread is 1 (Thread 0x7f7792f36740 (LWP 13097))]
(gdb) bt
#0  0x00007f7790c802a6 in libusb_submit_transfer () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#1  0x00007f7790c828c2 in libusb_control_transfer () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#2  0x00007f77927652fe in librealsense::platform::libuvc_uvc_device::get_data_usb (this=0x24b66d0, action=UVC_GET_MIN, control=2, unit=0)
    at librealsense/src/libuvc/libuvc.cpp:555
#3  0x00007f7792765879 in librealsense::platform::libuvc_uvc_device::get_pu_range (this=0x24b66d0, option=RS2_OPTION_BRIGHTNESS)
    at librealsense/src/libuvc/libuvc.cpp:610
#4  0x00007f77927617a4 in librealsense::platform::retry_controls_work_around::get_pu_range (this=0x24f4d10, opt=RS2_OPTION_BRIGHTNESS)
    at librealsense/src/backend.h:549
#5  0x00007f77922b7a48 in librealsense::platform::multi_pins_uvc_device::get_pu_range (this=0x250cc40, opt=RS2_OPTION_BRIGHTNESS)
    at librealsense/src/backend.h:805
#6  0x00007f77923ec5cc in librealsense::uvc_pu_option::<lambda(librealsense::platform::uvc_device&)>::operator()(librealsense::platform::uvc_device &) const (__closure=0x7ffd3218b930, dev=...)
    at librealsense/src/option.cpp:64
#7  0x00007f77923edb08 in librealsense::uvc_sensor::invoke_powered<librealsense::uvc_pu_option::get_range() const::<lambda(librealsense::platform::uvc_device&)> >(librealsense::uvc_pu_option::<lambda(librealsense::platform::uvc_device&)>) (this=0x24e9da0, action=...) at librealsense/src/sensor.h:200
#8  0x00007f77923ec630 in librealsense::uvc_pu_option::get_range (this=0x24e7550) at librealsense/src/option.cpp:64
#9  0x00007f7792392c6c in librealsense::uvc_sensor::try_register_pu (this=0x24e9da0, id=RS2_OPTION_BRIGHTNESS) at librealsense/src/sensor.cpp:698
#10 0x00007f77922d1857 in librealsense::platform_camera::platform_camera (this=0x24b6560, ctx=std::shared_ptr (count 7, weak 1) 0x24b6080, uvc_infos=std::vector of length 8, capacity 8 = {...}, group=..., 
    register_device_notifications=false, __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at librealsense/src/context.cpp:272
#11 0x00007f7792349eba in __gnu_cxx::new_allocator<librealsense::platform_camera>::construct<librealsense::platform_camera, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group, bool&>(librealsense::platform_camera*, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group&&, bool&) (this=0x7ffd3218c0b7, __p=0x24b6560) at /usr/include/c++/5/ext/new_allocator.h:120
#12 0x00007f779233e8be in std::allocator_traits<std::allocator<librealsense::platform_camera> >::construct<librealsense::platform_camera, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group, bool&>(std::allocator<librealsense::platform_camera>&, librealsense::platform_camera*, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group&&, bool&) (__a=..., __p=0x24b6560)
    at /usr/include/c++/5/bits/alloc_traits.h:530
#13 0x00007f7792334f89 in std::_Sp_counted_ptr_inplace<librealsense::platform_camera, std::allocator<librealsense::platform_camera>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group, bool&>(std::allocator<librealsense::platform_camera>, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group&&, bool&) (this=0x24b6550, __a=...)
    at /usr/include/c++/5/bits/shared_ptr_base.h:522
#14 0x00007f7792328e70 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<librealsense::platform_camera, std::allocator<librealsense::platform_camera>, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group, bool&>(std::_Sp_make_shared_tag, librealsense::platform_camera*, std::allocator<librealsense::platform_camera> const&, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group&&, bool&) (this=0x7ffd3218c338, __a=...) at /usr/include/c++/5/bits/shared_ptr_base.h:617
#15 0x00007f779231cb87 in std::__shared_ptr<librealsense::platform_camera, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<librealsense::platform_camera>, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group, bool&>(std::_Sp_make_shared_tag, std::allocator<librealsense::platform_camera> const&, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group&&, bool&) (
    this=0x7ffd3218c330, __tag=..., __a=...) at /usr/include/c++/5/bits/shared_ptr_base.h:1097
#16 0x00007f779230df10 in std::shared_ptr<librealsense::platform_camera>::shared_ptr<std::allocator<librealsense::platform_camera>, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group, bool&>(std::_Sp_make_shared_tag, std::allocator<librealsense::platform_camera> const&, std::shared_ptr<librealsense::context>&, std::vector<librealsense::platform::uvc_device_info, std::allocator<librealsense::platform::uvc_device_info> > const&, librealsense::platform::backend_device_group&&, bool&) (this=0x7ffd3218c330, __tag=..., __a=...)
    at /usr/include/c++/5/bits/shared_ptr.h:319
[... cut ...]
#22 0x000000000044ab7a in rs2::device_list::operator[] (this=0x7ffd3218c890, index=0) at librealsense/include/librealsense2/hpp/rs_device.hpp:219
#23 0x000000000044604c in main (argc=1, argv=0x7ffd3218d2c8) at /home/atritoman/development/librealsense/tools/enumerate-devices/rs-enumerate-devices.cpp:212
(gdb) 
```

1- Could someone provide some guidance, or just explain if indeed it is possible to distribute the drivers without depending on patching the kernel?

2- What udev permissions do I need to set ?

3- Do I still need admin to add udev rules/permissions to use user level drivers ?

Thanks in advance.

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
Hi @mjscosta,

Please try below to see if this can resolve the USB access permission issue.
Install Intel Realsense permission scripts located in librealsense source directory:
`$ sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`
`$ sudo udevadm control --reload-rules && udevadm trigger`
`reboot the system`
`Execute rs-enumerate-devices`

mjscosta, 2018-05-28:
1- I've installed the packages below, and avoided installing the dkms packages, and it works well for D415 camera. What I've noticed was that the executable rs-enumerate-devices  compiled with source using `cmake -DFORCE_LIBUVC=ON -DUSE_SYSTEM_LIBUSB=ON ../` still ends up having a core dump, but the one from the packages (librealsense2-utils) is not. Also there is no additional rules added to the udev.

```
ii  librealsense2:amd64                         2.11.0-0~realsense0.69                       amd64        Intel(R) RealSense(tm) Cross Platform API - runtime
ii  librealsense2-udev-rules:amd64              2.11.0-0~realsense0.69                       amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules
ii  librealsense2-utils:amd64                   2.11.0-0~realsense0.69                       amd64        Intel(R) RealSense(tm) Camera Capture API - utils and demos
```

2- Afterwards I've applied the method mentioned above, and copy the rules for udev, and after executing rs-enumerate-devices, I still have the same error, maybe I have to be in some special user group that can access the device, cause I've tested using sudo, and with sudo rs-enumerate-devices, it outputs the properly ending with a core dump, check logs below.

3- in summa:
3.1 - packaged rs-enumerate-devices, output ok
3.2 - compiled from source without having udev rules config/99-realsense-libusb.rules, error described above
3.3 - compiled from source without with udev rules config/99-realsense-libusb.rules, no error for D415 camera, but seems to have an error accessing some device (the platform camera?).
3.4 - compiled from source without with udev rules config/99-realsense-libusb.rules, and executed with **sudo**, no error for D415 camera, seems to output the platform device and ends up with an core dump.

3.3 - log:
```
$ ./rs-enumerate-devices 
 Device info: 
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	739112060767
    Firmware Version              : 	05.08.15.00
    Recommended Firmware Version  : 	05.09.09.02
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 1	  1920x1080	@ 25Hz	Y16
...
    Depth	  424x240	@ 6Hz	Z16

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color	  1920x1080	@ 30Hz	Y16
...
    Color	  320x180	@ 6Hz	YUYV

 28/05 14:12:19,428 ERROR [140132115003200] (types.h:179) Could not open device. Last Error: Permission denied
Segmentation fault (core dumped)
```

3.4 log:
```
$ sudo ./rs-enumerate-devices 
 Device info: 
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	739112060767
    Firmware Version              : 	05.08.15.00
    Recommended Firmware Version  : 	05.09.09.02
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 1	  1920x1080	@ 25Hz	Y16
...
    Depth	  424x240	@ 6Hz	Z16

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color	  1920x1080	@ 30Hz	Y16
...
    Color	  320x180	@ 6Hz	YUYV

unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
 28/05 14:17:25,476 ERROR [139958723729216] (types.h:179) Could not open device. Last Error: Device or resource busy
 Device info: 
    Name                          : 	Platform Camera
    Serial Number                 : 	2-7-6
    Physical Port                 : 	
    Product Id                    : 	B3ED

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
Segmentation fault (core dumped)
```

Any clue of what's going on ?

RealSense-Customer-Engineering, 2018-05-29:
[Realsense Customer Engineering Team Comment]
 Hi @mjscosta,

 1. The `libuvc` backend provide is for you to work with RealSense on **any** linux based system without patches using user-mode driver, but because it is a user-mode solution it comes with some stability and reliability limitations. Suggest to use LIBUSB instead, just run $ cmake -DFORCE_LIBUVC=OFF -DUSE_SYSTEM_LIBUSB=ON ../
The pre-built package (librealsense2-utils) should use LIBUSB, not LIBUVC.

2. I also suggest to upgrade your firmware to the latest (5.9.11) which resolves some issues. And, it looks your have another USB platform camera there. You can try to disable/remove it first and see how it works.

mjscosta, 2018-05-30:
Hi,

1- I understand the usage of LIBUVC, as it was suggested by @dorodnic, exactly to address the issue having a portable application across Linux, avoiding dkms modules. I still have some questions that are not clear to me on that regard.
1.1 - Which kernel versions are supported out of the out-of-the-box by librealsense2 (for D400 series) ?
1.2 - Which kernel versions are supported out of the out-of-the-box by librealsense2 (for SR300) ?
1.3 - Which backend is being used in windows for librealsense2 ?

2. I'll update the firmware ASAP, regarding the platform camera, it cannot be removed, as it is the inbuilt laptop camera.

Best,
mc

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @mjscosta,

Is the issue resolved?
By the way, SR300 support since kernel 4.12.

mjscosta, 2018-06-19:
yes, it is resolved. thanks.

closed 2018-06-19 by mjscosta