realsenseai/librealsense #1758: 403 Forbidden on RealSense repository

dheera, 2018-05-23:
403 error if one follows these instructions:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

[image]

dheera, 2018-05-23:
[image]

ev-mp, 2018-05-23:
Hello @dheera ,
The debian packages are currently supported for [Ubuntu 16 LTS](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution). It seem that you're trying to install it from Ubuntu 18 though.
Please follow the instructions and build the SDK from source code.  (related to #1670 )

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @dheera,

Did you follow the instruction to build from source?

dorodnic, 2018-07-26:
[Instructions](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages) has been updated to support Ubuntu 18

closed 2018-07-27 by dheera