realsenseai/librealsense #1760: Camera output not available when using it through ROS launch file

msieb1, 2018-05-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 14 | 
| Kernel Version (Linux Only)    |  4.4.0-112-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.11.0                 | 

### Issue Description

I launch the realsense2_camera rs_camera.launch node in my Baxter ROS environment, and get the following terminal out/error (the camera won't publish a topic anymore). Using realsense-viewer works, and the camera has been working before in ROS - this issue has just recently come up.

Output:

core service [/rosout] found
process[camera/realsense2_camera_manager-1]: started with pid [17536]
process[camera/realsense2_camera-2]: started with pid [17537]
[ INFO] [1527123271.151310827]: Initializing nodelet with 8 worker threads.
[ INFO] [1527123271.186376792]: RealSense ROS v2.0.3
[ INFO] [1527123271.186425401]: Running with LibRealSense v2.11.0
[ INFO] [1527123271.483263688]: getParameters...
[ INFO] [1527123271.616516995]: setupDevice...
[ INFO] [1527123271.616540586]: JSON file is not provided
[ INFO] [1527123271.616569570]: ROS Node Namespace: camera
[ INFO] [1527123271.616587929]: Device Name: Intel RealSense D435
[ INFO] [1527123271.616610393]: Device Serial No: 801212070655
[ INFO] [1527123271.616619389]: Device FW version: 05.08.15.00
[ INFO] [1527123271.616626313]: Device Product ID: 0x0B07
[ INFO] [1527123271.616632878]: Enable PointCloud: On
[ INFO] [1527123271.616642148]: Align Depth: Off
[ INFO] [1527123271.616660266]: Sync Mode: On
[ INFO] [1527123271.616682436]: Device Sensors: 
[ INFO] [1527123271.616703992]: Stereo Module was found.
[ INFO] [1527123271.616725738]: RGB Camera was found.
[ INFO] [1527123271.616748000]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1527123271.616767318]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1527123271.616779619]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1527123271.616790662]: setupPublishers...
[ INFO] [1527123271.623651521]: Expected frequency for depth = 30.00000
[ INFO] [1527123271.725285631]: Expected frequency for infra1 = 30.00000
[ INFO] [1527123271.806310280]: Expected frequency for infra2 = 30.00000
[ INFO] [1527123271.886645210]: Expected frequency for color = 30.00000
[ INFO] [1527123271.969029790]: setupStreams...
 23/05 20:54:31,969 WARNING [139955246512064] (sensor.cpp:330) Unregistered Media formats : [ UYVY Y8   ]; Supported: [ ]
[ INFO] [1527123271.970016117]: depth stream is enabled - width: 640, height: 360, fps: 30
[ INFO] [1527123271.970098430]: infra1 stream is enabled - width: 640, height: 360, fps: 30
[ INFO] [1527123271.970173075]: infra2 stream is enabled - width: 640, height: 360, fps: 30
 23/05 20:54:31,970 WARNING [139955246512064] (backend-v4l2.cpp:1156) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 23/05 20:54:31,970 WARNING [139955246512064] (sensor.cpp:330) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1527123271.971501166]: color stream is enabled - width: 640, height: 360, fps: 30
[ INFO] [1527123272.027314581]: publishStaticTransforms...
[ INFO] [1527123272.027509461]: RealSense Node Is Up!
 23/05 20:54:32,210 WARNING [139954152720128] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1527123272.211426563]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 23/05 20:54:32,252 WARNING [139954142476032] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 23/05 20:54:32,446 ERROR [139955246512064] (types.h:179) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ WARN] [1527123272.446821743]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
[ INFO] [1527123471.214508669]: Interrupt Signal is received! Terminating RealSense Node...
shutting down processing monitor...
... shutting down processing monitor complete
done
arpit@littleone:~/ros_wss/catkin_ws$ realsense-viewer
 23/05 20:58:02,233 WARNING [140180557534976] (types.cpp:57) get_pu(id=4) failed! Last Error: Input/output error
 23/05 20:58:02,286 WARNING [140180557534976] (sensor.cpp:330) Unregistered Media formats : [ UYVY Y8   ]; Supported: [ ]
 23/05 20:58:02,288 WARNING [140180557534976] (backend-v4l2.cpp:1156) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 23/05 20:58:02,288 WARNING [140180557534976] (sensor.cpp:330) Unregistered Media formats : [ RW16 ]; Supported: [ ]

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
Hi @msieb1,

I didn't see this issue on my test configuration is Ubuntu 16.04 + Kernel 4.13.0-43 + librealsense 2.11.1 + ROS wrapper 2.0.3. And, my firmware is 5.9.11.
You can try to upgrade your firmware to 5.9.11 to see if the same issue still there.

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
Hi @msieb1,

Any update? Still saw this issue on your system?

msieb1, 2018-06-14:
I haven't investigated further, but at some point it just worked again. Thanks for the help.