realsenseai/librealsense #1761: RealSense D415 RGB shows no frame in Intel RealSense Viewer

164579504, 2018-05-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |Win10| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)   | 
| Platform                                 | PC|
| SDK Version                            |  RealSense SDK 2.0                         | 

### Issue Description
Hi all,
Things happened when I use Intel RealSense Viewer for capturing RGB frame, messages on the screen shows "no frame". 
My device is D415, and I was just upgrade its firmware to 05.09.11.00. My USB3.0 port is Renesas Electronics USB 3.0 Host Controller. The USB data cable is exactly usb 3 data cable, but  is 1.5m long.(I wonder if I should use this long cable, but in my application such length is necessary). I will post some pictures on for more intuitive understanding of my problems. The problem is surely more than this one.
[image]
Fig1. The devices of mine

[image]
Fig2. When I trying to use 'Depth Quality tool', the software goes crash.

[image]
Fig3. When I click the RGB button to open the RGB channel, it shows no frame received

dorodnic, 2018-05-26:
Hi @164579504 
We will try to assist to find the problem. There is nothing immediately noticeably wrong with the setup, but it is certainly not a known issue. Could you please share the exact Windows build number? (using system info) 
Did you try the same camera on another Windows PC, just to rule out hardware problem?

164579504, 2018-05-28:
hi,thank u for help.
I got three info about my OS. 
1. Win 10 pro
2. 1709
3. 16299.15
Would these be helped?

I just try my friends' PC, his USB 3.0 interface belongs to INTEL co. then amazing happened that all the things goes well than ever before. The RGB stream appeared and no any crash of the software occured. Is it because the USB 3.0 interface cooperation?

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
Hello @164579504,

The fact that your camera works on another PC indicates that the problem is most likely with the USB controller of your target PC. To confirm that the problem is with your particular PC, test the camera on more than one PC.

Regards,
Jesus G.
Intel Customer Support

closed 2018-07-02 by RealSense-Customer-Engineering