realsenseai/librealsense #1764: Realsense in browsers?

jb455, 2018-05-24:
Hi,

Is it possible to get Realsense cameras (including the R400 series) streaming in browsers with accessible depth data etc? I was looking at the node.js wrapper, assuming that was for browser compatibility, but as far as I can tell that's just for server-side, or at least capturing on server and streaming data to a client - is that correct?
I've also found a [W3C repository](https://github.com/w3c/mediacapture-depth/) for developing depth in-browser functionality, but it looks like that's been abandoned now.
Is anyone aware of a way to get this working currently, or if a solution is actively being worked on anywhere?

Thanks!

dorodnic, 2018-05-29:
Hi @jb455 
Currently we don't have a solution for in-browser as far as I know (@tingshao - are you aware of any solution?)
Running node.js server is probably the easiest way to integrate with a web App.

tingshao, 2018-05-30:
Hi @dorodnic  @jb455 
Chrome has partly supported the [w3c media capture depth extension](https://www.w3.org/TR/mediacapture-depth/) spec and realsense was among the first supported devices.  and you could try [a demo](https://github.com/01org/depth-camera-web-demo) using the realsense camera, I tried it using RS435, it worked.

RealSense-Customer-Engineering, 2018-06-14:
[Realsense Customer Engineering Team Comment]
@jb455
does suggestion from tingshao work for you?

jb455, 2018-06-14:
Yes thanks, I'll keep an eye on that repo to see if any further development is done.

closed 2018-06-14 by jb455