realsenseai/librealsense #1766: The depth map of intelD435 and the edge of RGB diagram can not correspond.

super-scyu, 2018-05-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-05-25:
Hi @super-scyu 
The D435 offers wide field of view for depth and IR data, but regular web-cam field of view for the RGB. As a result you only have RGB data on a subset of available depth.

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
hi @super-scyu,

do you need more support on this topic?

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
let me close ticket here, please feel free to open new ticket if any question.

closed 2018-06-21 by RealSense-Customer-Engineering