realsenseai/librealsense #1767: Auto exposure too dark

baptiste-mnh, 2018-05-24:


agrunnet, 2018-05-24:
Two answers to that: 
1. You can adjust the autoexposure setpoint under the advanced controls. 
2. In general You get better depth for images that look a bit dark. You should consider possibly just muktiplying the color image by a fixed number to make it brighter for your usage.

baptiste-mnh, 2018-05-24:
I got this error when I'm trying to modify the AE set point : 
`300 is out of bounds [400, 400]`

xtrawurst, 2018-05-25:
As a workaround to the "out of bounds" issue you can to use one of the presets and store it under a separate JSON file. Then you can edit the set point in the JSON file and load it up again.

A higher set point will result in a brighter IR image. I have good experiences with values in the 1000 - 1500 range.

Btw. from the screen shot it looks like you are way below the min range of the sensor. Hence the jumbled up depth image.

baptiste-mnh, 2018-05-25:
Thanks for the tips @xtrawurst !
I made a bot to test and export the IR image at different exposition. I will determine the best render in function of the light.

closed 2018-05-25 by baptiste-mnh