realsenseai/librealsense #1774: Python wrapper, pointcloud as numpy array has numpy.voids in it!

ralphbaertschi, 2018-05-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC/  |
| SDK Version                            |  { legacy  }                          | 

### Issue Description

I slightly change the python example saving a ply file:
I want to make a python-"realtime" viewer of the point cloud based on open3d.
Now the problem is that the numpy array I get from the pointcloud has numpy.voids in it.
I only get rid of those by using the command tolist and casting back to numpy.array.
Now this procedure is like a factor 100 slower than normal numpy operations.
Therefore, I only would get 5-Hz framerate on a possible viewer.

` points = pc.calculate(depth)
    pc.map_to(color)
    print("Saving to %d.ply..." % ind)
    #points.export_to_ply("data/%d.ply" % ind, color) 
    vtx = np.asanyarray(points.get_vertices())
    print(vtx.dtype)
    print(type(vtx),type(vtx[0]))
    vtx2 = np.array(vtx)
    print(vtx2.dtype)
    #vtx2 = vtx.astype(dtype=[('f','f','f')])
    t0 = time.time()
    vtx2 = vtx.tolist()
    vtx2 = np.array(vtx2)
    print(time.time() -t0)
    raise "s"`

Printout
Saving to 0.ply...
[('f0', '<f4'), ('f1', '<f4'), ('f2', '<f4')]
<class 'numpy.ndarray'> <class 'numpy.void'>
[('f0', '<f4'), ('f1', '<f4'), ('f2', '<f4')]
0.24765872955322266

0.2 seconds this is slower than snail  time.

ralphybbbaer, 2018-05-26:
I can use the pointcloud generation in open3d as in: https://github.com/IntelVCL/Open3D/issues/374

RealSense-Customer-Engineering, 2018-06-06:
[Realsense Customer Engineering Team Comment]
hi @ralphybbbaer,

this is interesting, so do you make it and is it still slow or not?
I want to see how it works too.

ralphybbbaer, 2018-06-06:
When postprocessing the depth image through librealsense and calculating the pointcloud with open3d and using the viewer in open3d I get an acceptable 10-20 Hz viewer in python.

closed 2018-06-06 by ralphybbbaer

baptiste-mnh, 2018-07-26:
Is there a way to get good fps using the pointscloud calculated by open3D converted in numpy?
The compute time to convert the numpy from the librealsense to the one compatible with open 3D is very long.

I've found this on github, have a look at this : 
https://github.com/baptiste-mnh/pySoRo#adding-two-dimensional-data-protocols
It explain how to speed up the compute time.

karlita101, 2021-04-08:
Hi there, I was also wondering if there may have been any new updates on improving the fps for real-time point cloud solutions.

4ri4Slr, 2021-06-16:
@karlita101  An easy fix for getting NumPy arrays without void dtypes is provided in #7629.