realsenseai/librealsense #1776: Android compile error

Performador, 2018-05-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Ubuntu | 
| Kernel Version (Linux Only)    |  4.9.0                                         | 
| Platform                                 | Android |
| SDK Version                            |  2.11.1                        | 

While following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/development/doc/android/AndroidNativeSamples.md), getting the following compile error.

Using android-ndk-r17.

Latest git hash: 95eba49e12d2256edeb15b9e5918bd9d74486fb0.

The Android toolchain c++ --version:

Android (4691093 based on r316199) clang version 6.0.2 (https://android.googlesource.com/toolchain/clang 183abd29fc496f55536e7d904e0abae47888fc7f) (https://android.googlesource.com/toolchain/llvm 34361f192e41ed6e4e8f9aca80a4ea7e9856f327) (based on LLVM 6.0.2svn)
Target: x86_64-unknown-linux-gnu
Thread model: posix

=======================================================
Various errors:

In file included from /tmp/librealsense/src/pipeline.cpp:10:
In file included from /tmp/librealsense/src/media/ros/ros_writer.h:16:
/tmp/librealsense/src/media/ros/ros_file_format.h:441:42: error: call to 'stoll' is ambiguous
            return static_cast<uint32_t>(std::stoll(id_str));

In file included from /tmp/librealsense/src/pipeline.cpp:10:
In file included from /tmp/librealsense/src/media/ros/ros_writer.h:16:
/tmp/librealsense/src/media/ros/ros_file_format.h:775:32: error: call to 'stoi' is ambiguous
                retval.index = std::stoi(index_str);
                               ^~~~~~~~~

In file included from /tmp/librealsense/src/pipeline.cpp:10:
In file included from /tmp/librealsense/src/media/ros/ros_writer.h:16:
/tmp/librealsense/src/media/ros/ros_file_format.h:789:29: error: call to 'stoi' is ambiguous
                int index = std::stoi(index_str);

/tmp/librealsense/src/media/ros/ros_file_format.h:848:54: error: call to 'stoll' is ambiguous
                sensor_index = static_cast<uint32_t>(std::stoll(ros_topic::get<3>(topic)));

In file included from /tmp/librealsense/src/pipeline.cpp:10:
/tmp/librealsense/src/media/ros/ros_writer.h:300:56: error: reference to overloaded function could not be resolved; did you mean to
      call it?
            frame_timestamp_msg.value = to_string() << std::hexfloat << pose->get_frame_timestamp();

bu3w, 2018-05-25:
comment out all code in common/android_helpers.h

Performador, 2018-05-26:
It compiles with common/android_helpers.h commented out.

RealSense-Customer-Engineering, 2018-05-28:
[Realsense Customer Engineering Team Comment]
Thanks, @Performador, @bu3w.
Need to further check internally about this compile bug.

dorodnic, 2018-05-29:
Well, what concerns me is that we do build Android in our CI, using the following script: [.travis.yml:125-131](https://github.com/IntelRealSense/librealsense/blob/master/.travis.yml#L125-L131)
Is it the version of the NDK that makes the difference? If so, can we conditionally include `common/android_helpers.h` for the old one?

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
@Performador  The pull request for this compile error is under review. Do you have any other questions about this issue? If not, we'll mark this solved. Thanks!

Performador, 2018-06-28:
Sounds good. Thanks.

closed 2018-06-29 by RealSense-Customer-Engineering