realsenseai/librealsense #1777: i am using the intel-D435,i do not know how to get the 16bit depth image ?i can only get the 8bit color image

super-scyu, 2018-05-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-05-26:
Hi @super-scyu 
What programming language are you using?

super-scyu, 2018-06-04:
i am using c++发自我的iPhone------------------ Original ------------------From: Sergey Dorodnicov  <notifications@github.com>Date: 周日,5月 27,2018 1:11 上午To: IntelRealSense/librealsense <librealsense@noreply.github.com>Cc: super-scyu <yushichao1991@nuaa.edu.cn>, Mention <mention@noreply.github.com>Subject: Re: [IntelRealSense/librealsense] i am using the intel-D435,i do notknow how to get the 16bit depth image ?i can only get the 8bit color image(#1777)Hi @super-scyu
What programming language are you using?

{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [IntelRealSense/librealsense] i am using the intel-D435,i do not know how to get the 16bit depth image ?i can only get the 8bit color image (#1777)",
"sections": [
{
"text": "",
"activityTitle": "**Sergey Dorodnicov**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@dorodnic",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"IntelRealSense/librealsense\",\n\"issueId\": 1777,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"IntelRealSense/librealsense\",\n\"issueId\": 1777\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/IntelRealSense/librealsense/issues/1777#issuecomment-392274977"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 339430452\n}"
}
],
"themeColor": "26292E"
}

dorodnic, 2018-06-04:
Something like:
```cpp
using namespace rs2;
pipeline p;
p.start();
auto fs = p.wait_for_frames();
auto depth = fs.get_depth_frame();
auto pixels = (uint16_t*)depth.get_data();
```

dorodnic, 2018-06-16:
Hope this helps, feel free to open more issues if you have questions

closed 2018-06-16 by dorodnic