realsenseai/librealsense #1778: disparity transform using python wrapper

adarvit, 2018-05-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400  | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |   2.11.1  | 

i am trying to use disparity transform with python according to c++ API, but without success
How can I use this filter?
`        self.disparity = rs.disparity_transform(True)
`
`self.disparity.update_transformation_profile(filtered)`

lramati, 2018-05-28:
Like in the C++ API, you create the filter with `disparity = pyrealsense2.disparity_transform()`.
The frames are then processed via `new_frame = disparity.process(old_frame)`

adarvit, 2018-05-28:
thanks, what is the meaning of those depth values after using this method?
[image]

dorodnic, 2018-05-28:
Usually this block is used before `spatial` and `temporal` filters and the "rolled-back" using another disparity transform with `rs.disparity_transform(False)`. 
If you want to use these values directly, you should cast data to `float`, and interpret the results as displacement of every pixel between left and right views, in pixels.

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @Iramati,

Is above answer resolve your issue?

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @adarvit,

Does above answer resolve your issue?

adarvit, 2018-06-20:
yes, thank you

closed 2018-06-20 by adarvit