realsenseai/librealsense #1779: multicam example is nor working on OSX

shiyamon, 2018-05-27:
It seems multicam example of librealsense2 is not working here.
After I build multicam example, the app just hangup. No error can be seen in Xcode.
A similar thing happens when I try to access res2::device inside device_list.

auto deviceList = _ctx->query_devices();
if(deviceList.size() > 0) {
 device = deviceList[0];
}

Am I missing something?

RealSense-Customer-Engineering, 2018-05-29:
[Realsense Customer Engineering Team Comment]
hi @shiyamon,

do you tested with other example? would like to know if only multicam example fail or all example failed.

dorodnic, 2018-05-31:
We are looking into multi-cam issues with MacOS. Will update once we have something.

RealSense-Customer-Engineering, 2018-06-06:
[Realsense Customer Engineering Team Comment]
hi @shiyamon,

maybe the same issue as #1586, can you please have a try with workaround there?
adding sleep(1); some cases need sleep(5)

shiyamon, 2018-06-06:
@RealSense-Customer-Engineering  Sorry for replying this late!
I haven't tested but that workaround should work. I found that no error occurs when I debug with breakpoints on each lines.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
hi @shiyamon,

Proper fix is now in the development branch, will be released soon

closed 2018-07-26 by dorodnic