realsenseai/librealsense #1781: intel realsense D435 did not work...

yeongwoonIm, 2018-05-28:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1)                                          | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  Intel RealSense SDK 2.0                         | 

### Issue Description
Hi, I'm using D435
I tested D435 on my computer. it was working very well. no problem happened.

but, I tested it on another computer that is for customer.

I installed those below
- Intel.RealSense.SDK
- visual C++ 2013 redistributable package
- visual C++ 2015 redistributable package
- visual C++ 2017 redistributable package
- librealsense-2.10.3

at first, my code didn't work. so I tried sample code for C#

librealsense-2.10.3\librealsense-2.10.3\wrappers\csharp\Intel.RealSense.2015

I ran cs-tutorial-1-depth. but 
[image]

this console message show. There are 1 connected RealSense devices.
and then freeze...

so I ran Intel RealSense Viewer.exe
[image]

that shows long time.. and then freeze...
[image]

What shuold I do??

What I have missed??

Please Help me..

I must test D435 and it has to work well!!!

My company wants depth cam, not Kinect

[image]

Device installed, I think...

// D435 가 제컴퓨터 에서는 잘 되는데 왜, 납품용 PC에서는 안되는건가요...
// 도와주세요...
//해결법좀 알려주세요..

RealSense-Customer-Engineering, 2018-05-28:
[Realsense Customer Engineering Team Comment]
Hi @yeongwoonlm,

Are the apps installed in below location? I guess it probably the OS language issue since we run those apps in English Win10 OS.
C:\Program Files (x86)\Intel RealSense SDK 2.0\tools
You can try to copy the 'bin' and 'tools' in C:\temp path and run "realsense-viewer" app again.
Please also run "rs-enumerate-devices" to show us the log.

yeongwoonIm, 2018-05-29:
it's working now

how I solve this issue is I disconnect other vision cameras, and other depth cameras(Cube depth). and uninstall devices and S/W.
disconnect intel realsense and uninstall device. 
- intel(R) RealSense(TM) Camera Color
- intel(R) RealSense(TM) Camera Depth

reboot PC

reconnect intel realsense.

wait install intel realsense device.

and then run "realsense-viewer" app.

[image]

it's working now~

and I ran "rs-enumerate-devices" . I can't find any logs.. in  "C:\Program Files (x86)\Intel RealSense SDK 2.0\tools"

RealSense-Customer-Engineering, 2018-05-29:
[Realsense Customer Engineering Team Comment]
Hi @yeongwoonlm,

Great to hear it can work. I guess this is due to other USB cameras affects to use realsense-viewer to operate D415 camera.
If you like to further check the issue related to other USB vision cameras, you can set below in the command window and use the same window to run the realsense-viewer. Then, you can find the log file in the same folder.
set LRS_LOG_LEVEL=DEBUG

By the way, when you run rs-enumerate-devices in the command window, like to see its the info it shows.

closed 2018-05-29 by yeongwoonIm