realsenseai/librealsense #1783: what is the output of get_vertices() exactly

zsert3, 2018-05-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00| 
| Operating System & Version |  Win 10 | 
| Platform                                 |  PC  |
| SDK Version                            |  SDK 2.10.?                          |

When you execute the function `points.get_vertices()` you wil get an pointer to an X, Y, Z coördinate.
This coordinate has the following type: `const rs2::vertex* name`

My question:
Normally if you do `name++` to a pointer you will get the next value.
This pointer refers to an array where all the coordinates of a frame refers, and the function `get_vertices()` gives the start of that array.

But I am not sure if I perform the ++ operation which coordinate I will get, for example:
The pointer I get is the top most left coordinate of a frame, if the ++ operation is performed will I then get the coordinate left from the starting point, or the coordinate below it, or do I see this all wrong?

dorodnic, 2018-05-28:
Hi @zsert3 
`points` frame behaves like the depth frame it was created from: It contains the array of stacked rows. 
`++` will bring you to the next point to the right, unless you reach of of row, then you will jump to start of next row. Some points are going to be `(0,0,0)`. We keep those to allow easy navigation within the point-cloud grid (you can access point corresponding to pixel (i, j) using `v[j * width() +i]`)

dorodnic, 2018-06-05:
I think this can be resolved, @zsert3 hope this answers your question, if you have any follow-up questions feel free to open more issues!

closed 2018-06-05 by dorodnic