realsenseai/librealsense #1785: Debian ARM board: 'INZI' is not natively supported ... please upgrade to kernel 4.12 or later.

KeCh96, 2018-05-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Debian  | 
| Kernel Version (Linux Only)    |  4.11.0 | 
| Platform                                 | ARM board |
| SDK Version                            |  SDK2  | 

I want to get depth frames and process by ARM board with Debian system. I have followed [this guide](https://github.com/IntelRealSense/librealsense/issues/980) and  build pyrealsense2.so in my ARM board. Then I run the following code in python:

> import pyrealsense2 as rs
> pipeline = rs.pipeline()
> pipeline.start()

I run into an error:

> Traceback (most recent call last):
>   File "test_realsense.py", line 5, in <module>
>     pipeline.start()
> RuntimeError: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
> Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device

So I tried to update my linux kernel, and use `aptitude search linux-image | awk '{print $2}'` to see all support kernels. I get the following result:

> linux-image-4.11.0-qcomlt
> linux-image-4.11.0-qcomlt-arm:armhf
> linux-image-4.11.0-qcomlt-arm64
> linux-image-4.11.0-qcomlt-arm64-dbg
> linux-image-4.11.0-qcomlt-dbg
> linux-image-4.14.0-qcomlt-arm:armhf
> linux-image-4.14.0-qcomlt-arm64
> linux-image-4.14.0-qcomlt-arm64-dbg
> linux-image-4.9.0-qcomlt
> linux-image-4.9.0-qcomlt-dbg

I download 4.14 kernel and use `apt-get purge linux-image-4.11.0-qcomlt` to remove the 4.11 kernel. Then I type `dpkg --get-selections |grep linux-image` and only see 4.14. But I type `uname -a` and find the kernel is still 4.11
I try the python code `pipeline.start()`  in this environment, and get "No device" error.

I also tried 4.9.0, and the result is exactly the same with 4.14.

I wonder if the camera will work in 4.10.0 or 4.4.0 kernel, but as the above information shows, my ARM board does not seem to support 4.10.0 or 4.4.0 kernel. Thus this solution seems not applicable for me.
 

My main goal to get depth frame, (and not some of more advanced controls / metadata support), I notice [this guide](https://github.com/IntelRealSense/librealsense/issues/1186), but I still confused what I should do in my condition. So I open this new issue. Could you please help me? @dorodnic  @zivsha

dorodnic, 2018-05-29:
Hi @KeCh96 
The same answer from #1186 still holds: you can either manually patch the kernel or run cmake with `-DFORCE_LIBUVC=true`.

KeCh96, 2018-05-31:
Thank you!! I run cmake with `-DFORCE_LIBUVC=true` and get depth frame successfully! @dorodnic 

After several seconds, I get 391 frames, but an `Segmentation fault` error occurs:

> 372
> 373
> 374
> 375
> 376
> 377
> 378
> 379
> 380
> 381
> 382
> 383
> 384
> 385
> 386
> 387
> 388
> 389
> 390
> 391
> Segmentation fault

I run my code again and get `Segmentation fault` immediately (I get no frame this time). 
Then I plug out SR300 and plug in it again, then run my code. After about 300 frames, the `Segmentation fault` error occurs again.

KeCh96, 2018-05-31:
According to [this guide](https://github.com/IntelRealSense/librealsense/issues/10), this error may caused by multiple instances of the same device on Linux. But I only use one SR300. @sgorsten 

According to [this issue](https://github.com/IntelRealSense/librealsense/issues/885), we can use the following code to avoid this bug:

>             if (
>                     !ir_frame || ir_frame.get_data() == NULL ||
>                     !color_frame || color_frame.get_data() == NULL ||
>                     !depth_frame || depth_frame.get_data() == NULL) {
> 
>                 --i;
>                 continue;
>             }
So I write my code as following:

> pipeline = rs.pipeline()
> pipeline.start()
> 
> i=0
> while True:
>     try:
>         frames = pipeline.wait_for_frames()
>         depth = frames.get_color_frame()
>         depth_data = depth.as_frame().get_data()
>         i=i+1
>         print i
>     except:
>         continue
But the `Segmentation fault` error still occurs.

KeCh96, 2018-05-31:
I found the following annotation in `/third_party/sqlite/sqlite3.h`. This information may be helpful.
```
**
** ^Calling sqlite3_free() with a pointer previously returned
** by sqlite3_malloc() or sqlite3_realloc() releases that memory so
** that it might be reused.  ^The sqlite3_free() routine is
** a no-op if is called with a NULL pointer.  Passing a NULL pointer
** to sqlite3_free() is harmless.  After being freed, memory
** should neither be read nor written.  Even reading previously freed
** memory might result in a segmentation fault or other severe error.
** Memory corruption, a segmentation fault, or other severe error
** might result if sqlite3_free() is called with a non-NULL pointer that
** was not obtained from sqlite3_malloc() or sqlite3_realloc().
**
```

KeCh96, 2018-06-02:
I solve this problem!!  Just by adding a swap on my ARM board. (my ARM board do not have swap natively)

KeCh96, 2018-06-04:
I open a [new issue](https://github.com/IntelRealSense/librealsense/issues/1821) on this problem. To analyze the reason of the error, I show my 'cmake' and 'make' output. Could someone provide some help?

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Hi @KeCh96,

Since you resolve this issue, still need any support for this topic?

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
If no other further c

closed 2018-08-06 by RealSense-Customer-Engineering