realsenseai/librealsense #1789: How to set HW sync for multicamera ?

hansonleelisong, 2018-05-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                 | 
| Firmware Version                   | (05.09.02.00)                                           | 
| Operating System & Version |   Win 10                                                 | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                  | PC                                                           |
| SDK Version                            | V2.10.1                                                   | 

### Issue Description
How to set HW sync for multicamera ?
i send the HW sync command by rs-terminal.exe as below:

<-----------------------
Waiting for RealSense device to connect...
RealSense device has connected...
Device Name: Intel RealSense 415
Device Path: \\?\usb#vid_8086&pid_0ad3&mi_00#6&711b82e&0&0000#{24e552d7-6523-47f7-a647-d3465bf1f5ca}\global
Device S/N: 751412060646

#>14 00 ab cd 64 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

e5 ff ff ff
---------------------->

It is always show e5 ff ff ff , and there is no any hw sync signal detected.

This command is from RealSense_Multiple_Camera_WhitePaper.pdf  chapter 3. A. HW sync command.
[RealSense_Multiple_Camera_WhitePaper.pdf](https://github.com/IntelRealSense/librealsense/files/2048127/RealSense_Multiple_Camera_WhitePaper.pdf)

dorodnic, 2018-05-29:
I believe you need to reach to [Intel support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) for a dedicated firmware, as this is not **yet** enabled in the public firmware. 
Perhaps @RealSense-Customer-Engineering can comment how to open a Zendesk ticket.

RealSense-Customer-Engineering, 2018-05-30:
[Realsense Customer Engineering Team Comment]
hi @hansonleelisong,

the FW is not ready for release yet.
as you mentioned this is urgent, would like to know your schedule and if you are NDA customer to Intel?
if so, please go to Zendesk for more direct support.
on the other hand will file request base on this.

hansonleelisong, 2018-05-30:
Hi sir,

Is there any plan or not about this function?  could you show the develop schedule to me ?  

Thanks.

agrunnet, 2018-06-20:
Intel is preparing to release new FW within a few weeks that will officially shoot the multicamera function. Sorry for the delay.

dorodnic, 2018-07-11:
[5.9.14](https://downloadcenter.intel.com/download/27954/Latest-Firmware-for-Intel-RealSense-D400-Product-Family) is **finally** out, allowing hardware sync between multiple D415 cameras. D435 is coming soon.

closed 2018-08-14 by RealSense-Customer-Engineering