realsenseai/librealsense #1793: realsense-viewer emitter auto option

jbvo, 2018-05-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.13.0-43                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.11.0-0~realsense0.69                        | 

### Issue Description

Through apt I installed the most recent versions of librealsense2 and its sibling packages and am trying to use them with a RealSense D415 connected to a USB3 port. Now when using realsense viewer, I observed that the checkbox `Emitter enabled` shows the following tooltip: `Power of the DS5 projector, 0 meaning projector off, 1 meaning projector on, 2 meaning projector in auto mode`. I'm wondering how I can use the auto mode, did I miss something?

dorodnic, 2018-05-30:
Hi @jbvo 
This seems to be related to `05.09.11.00` firmware. We are checking if it is a bug or intentional change.

RealSense-Customer-Engineering, 2018-06-01:
[Realsense Customer Engineering Team Comment]
Hi @jbvo,
the option 2 is deprecated,  but you can use your own rules to toggle the projector on/off
the emitter (laser projector) is to project invisible IR pattern that can help increase the depth accuracy especially on area that does not have much features such as white wall
you can refer to below link has some guidance about how to get good depth image
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

best

closed 2018-06-27 by RealSense-Customer-Engineering

wokanmanhua, 2021-11-11:
> [Realsense Customer Engineering Team Comment] Hi @jbvo, the option 2 is deprecated, but you can use your own rules to toggle the projector on/off the emitter (laser projector) is to project [...]

What does "emitter always on" mean?