realsenseai/librealsense #1798: Pixel 2 Android unable to connect

Performador, 2018-05-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Ubuntu | 
| Kernel Version (Linux Only)    |  4.9.0                                         | 
| Platform                                 | Android |
| SDK Version                            |  2.11.1                        | 

While following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/development/doc/android/AndroidNativeSamples.md), getting the following runtime error (unable to see the device).

Using android-ndk-r17.

Latest git hash: 95eba49e12d2256edeb15b9e5918bd9d74486fb0.

lsusb:
```
Bus 001 Device 003: ID 8086:0ad3
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0003
```

When plugging the device, I get the following in dmesg:
```
[ 1851.754045] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[ 1851.754117] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[ 1851.754127] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[ 1851.754134] usb usb2: Product: xHCI Host Controller
[ 1851.754142] usb usb2: Manufacturer: Linux 4.4.88-g3acf2d53921d xhci-hcd
[ 1851.754149] usb usb2: SerialNumber: xhci-hcd.0.auto
[ 1851.759308] hub 2-0:1.0: USB hub found
[ 1851.759349] hub 2-0:1.0: 1 port detected
[ 1852.956869] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[ 1853.078816] usb 1-1: New USB device found, idVendor=8086, idProduct=0ad3
[ 1853.078842] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1853.078850] usb 1-1: Product: Intel(R) RealSense(TM) 415
[ 1853.078857] usb 1-1: Manufacturer: Intel(R) RealSense(TM) 415
```

It seems like it's using USB2 interface. Is this a deal killer?

```
./rs-enumerate-devices                                                                                                                  
No device detected. Is it plugged in?

./rs-depth
There are 0 connected RealSense devices.
```

RealSense-Customer-Engineering, 2018-05-31:
[Realsense Customer Engineering Team Comment]
Hi @Performador,

It should work even only USB2 supported. FW 5.9.11 can support USB2.
It is a 'root' device, right? What's your CMAKE options? Is "-DFORCE_LIBUVC=TRUE" used?

Performador, 2018-05-31:
Thanks, I did not have -DFORCE_LIBUVC=TRUE in my cmake line.

Filed: https://github.com/IntelRealSense/librealsense/pull/1808 to update the AndroidNativeSamples.md since it does not work without this.

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
Hi @Performador,

I will close this since this issue is resolved with the correct CMAKE option.

closed 2018-06-12 by RealSense-Customer-Engineering

Performador, 2018-06-12:
Not sure if this is resolved. With libuvc the samples are able to see the device on Android but are not receiving data.

For example, rs-depth will get stuck after showing the connected device:
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D415
    Serial number: 810512061144
    Firmware version: 05.09.11.00

rs-enumerate-devices seems to correctly enumerate the devices.

Without libuvc the samples will not even see the connected device.

Performador, 2018-06-12:
Can we please re-open this bug. 

Right now it is not possible to use D415 on Android O.

PR #1808 (2018-06-03): Update AndroidNativeSamples.md
The SDK does not seem to find the device without libuvc on Android devices.

Fixes: #1798