realsenseai/librealsense #1799: Program terminated with signal SIGSEGV, Segmentation fault.

omerbrandis, 2018-05-31:
hello, 

i've written a program that uses two d415 cameras using librealsense 2.10.1

and i'm getting the following error on program exit 
Program terminated with signal SIGSEGV, Segmentation fault.

i've generated a core dump, and used "gdb" to analyze it, 

here is the back trace:
#0  __GI___pthread_mutex_lock (mutex=0x8) at ../nptl/pthread_mutex_lock.c:67
67	../nptl/pthread_mutex_lock.c: No such file or directory.
[Current thread is 1 (Thread 0x7f08b97fa700 (LWP 11379))]
(gdb) bt
#0  __GI___pthread_mutex_lock (mutex=0x8) at ../nptl/pthread_mutex_lock.c:67
#1  0x00007f09c6a4d7c3 in __gthread_mutex_lock (__mutex=<optimized out>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:748
#2  __gthread_recursive_mutex_lock (__mutex=<optimized out>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:810
#3  std::recursive_mutex::lock (this=<optimized out>) at /usr/include/c++/5/mutex:176
#4  0x00007f09c6a4aad9 in std::lock_guard<std::recursive_mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/5/mutex:386
#5  el::base::RegisteredLoggers::get (this=0x0, id="librealsense", forceCreation=<optimized out>)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:1805
#6  0x00007f09c6a4af59 in el::base::Writer::initializeLogger (this=this@entry=0x7f08b97f9610, loggerId="librealsense", lookup=lookup@entry=true, needLock=needLock@entry=true)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:2432
#7  0x00007f09c6a4b3c2 in el::base::Writer::construct (this=this@entry=0x7f08b97f9610, count=count@entry=1, loggerIds=loggerIds@entry=0x7f09c6a777bd "librealsense")
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:2424
#8  0x00007f09c683c8e8 in librealsense::uvc_sensor::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)>::operator()(librealsense::platform::frame_object, std::function<void()>) (__closure=0x133fc1c0, f=..., continuation=..., p=...) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/sensor.cpp:421
#9  0x00007f09c683da4e in std::_Function_handler<void(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>), librealsense::uvc_sensor::open(const stream_profiles&)::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)> >::_M_invoke(const std::_Any_data &, <unknown type in /usr/lib/debug/.build-id/1a/6e1cad74f642809e8b753ce28d8c061357ee4b.debug, CU 0x7d0916, DIE 0x8e0a45>, <unknown type in /usr/lib/debug/.build-id/1a/6e1cad74f642809e8b753ce28d8c061357ee4b.debug, CU 0x7d0916, DIE 0x8e0a4a>, <unknown type in /usr/lib/debug/.build-id/1a/6e1cad74f642809e8b753ce28d8c061357ee4b.debug, CU 0x7d0916, DIE 0x8e0a4f>) (__functor=..., __args#0=<optimized out>, __args#1=<optimized out>, __args#2=<optimized out>)
    at /usr/include/c++/5/functional:1871
#10 0x00007f09c68b26e2 in std::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) const (__args#2=..., __args#1=..., __args#0=..., this=0x136e6480) at /usr/include/c++/5/functional:2267
#11 librealsense::platform::v4l_uvc_device::poll (this=this@entry=0x136e6370) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/linux/backend-v4l2.cpp:804
#12 0x00007f09c68b2a58 in librealsense::platform::v4l_uvc_device::capture_loop (this=0x136e6370)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/linux/backend-v4l2.cpp:1180
#13 0x00007f09ac749510 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#14 0x00007f09aca1c6ba in start_thread (arg=0x7f08b97fa700) at pthread_create.c:333
#15 0x00007f09abeaa41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

please advise.
Omer.

dorodnic, 2018-05-31:
Hi @omerbrandis 
Did you try 2.11.x? I believe this problem was resolved in the later release.

omerbrandis, 2018-06-07:
hello, 

i have now upgraded to the latest release , 5.12 , 
this did not solve the problem.

here is the new back trace:

#0  __GI___pthread_mutex_lock (mutex=0x8) at ../nptl/pthread_mutex_lock.c:67
67	../nptl/pthread_mutex_lock.c: No such file or directory.
[Current thread is 1 (Thread 0x7fc6037fe700 (LWP 7348))]
(gdb) bt
#0  __GI___pthread_mutex_lock (mutex=0x8) at ../nptl/pthread_mutex_lock.c:67
#1  0x00007fc6f72f5913 in __gthread_mutex_lock (__mutex=<optimized out>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:748
#2  __gthread_recursive_mutex_lock (__mutex=<optimized out>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:810
#3  std::recursive_mutex::lock (this=<optimized out>) at /usr/include/c++/5/mutex:176
#4  0x00007fc6f72f2c29 in std::lock_guard<std::recursive_mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/5/mutex:386
#5  el::base::RegisteredLoggers::get (this=0x0, id="librealsense", forceCreation=<optimized out>)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:1805
#6  0x00007fc6f72f30a9 in el::base::Writer::initializeLogger (this=this@entry=0x7fc6037fd610, loggerId="librealsense", lookup=lookup@entry=true, needLock=needLock@entry=true)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:2432
#7  0x00007fc6f72f3512 in el::base::Writer::construct (this=this@entry=0x7fc6037fd610, count=count@entry=1, loggerIds=loggerIds@entry=0x7fc6f731f94b "librealsense")
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:2424
#8  0x00007fc6f70d4b70 in librealsense::uvc_sensor::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)>::operator()(librealsense::platform::frame_object, std::function<void()>) (__closure=0x29db8c0, f=..., continuation=..., p=...) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/sensor.cpp:425
#9  0x00007fc6f70d5d1e in std::_Function_handler<void(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>), librealsense::uvc_sensor::open(const stream_profiles&)::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)> >::_M_invoke(const std::_Any_data &, <unknown type in /usr/lib/debug/.build-id/a8/9e6f6a5aa943744cf9053b49748a64d3a61ec1.debug, CU 0x7fed27, DIE 0x912e77>, <unknown type in /usr/lib/debug/.build-id/a8/9e6f6a5aa943744cf9053b49748a64d3a61ec1.debug, CU 0x7fed27, DIE 0x912e7c>, <unknown type in /usr/lib/debug/.build-id/a8/9e6f6a5aa943744cf9053b49748a64d3a61ec1.debug, CU 0x7fed27, DIE 0x912e81>) (__functor=..., __args#0=<optimized out>, __args#1=<optimized out>, __args#2=<optimized out>)
    at /usr/include/c++/5/functional:1871
#10 0x00007fc6f714d506 in std::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) const (__args#2=..., __args#1=..., __args#0=..., this=0x2953b40) at /usr/include/c++/5/functional:2267
#11 librealsense::platform::v4l_uvc_device::poll (this=this@entry=0x2953a20) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/linux/backend-v4l2.cpp:862
#12 0x00007fc6f714d858 in librealsense::platform::v4l_uvc_device::capture_loop (this=0x2953a20) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/linux/backend-v4l2.cpp:1238
#13 0x00007fc6dcfc7510 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#14 0x00007fc6dd29a6ba in start_thread (arg=0x7fc6037fe700) at pthread_create.c:333
#15 0x00007fc6dc72841d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

please advise,
Omer.

omerbrandis, 2018-06-07:
i've changed realsense's logging level to None, 
but error persists,
back trace is a bit different than before ( still looks like it's related to easylogging++)

#0  __GI___pthread_mutex_lock (mutex=0x8) at ../nptl/pthread_mutex_lock.c:67
#1  0x00007f6fbe39d913 in __gthread_mutex_lock (__mutex=<optimized out>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:748
#2  __gthread_recursive_mutex_lock (__mutex=<optimized out>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:810
#3  std::recursive_mutex::lock (this=<optimized out>) at /usr/include/c++/5/mutex:176
#4  0x00007f6fbe39ac29 in std::lock_guard<std::recursive_mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/5/mutex:386
#5  el::base::RegisteredLoggers::get (this=0x0, id="librealsense", forceCreation=<optimized out>)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:1805
#6  0x00007f6fbe39b0a9 in el::base::Writer::initializeLogger (this=this@entry=0x7f6ec7ffe350, loggerId="librealsense", lookup=lookup@entry=true, needLock=needLock@entry=true)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:2432
#7  0x00007f6fbe39b512 in el::base::Writer::construct (this=this@entry=0x7f6ec7ffe350, count=count@entry=1, loggerIds=loggerIds@entry=0x7f6fbe3c794b "librealsense")
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/third-party/easyloggingpp/src/easylogging++.cc:2424
#8  0x00007f6fbe0ffae5 in librealsense::frame::log_callback_start (this=<optimized out>, timestamp=1528367972606.4148)
    at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/archive.cpp:468
#9  0x00007f6fbe245113 in librealsense::frame_source::invoke_callback (this=0x239d1fa8, frame=...) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/source.cpp:117
#10 0x00007f6fbe17d163 in librealsense::uvc_sensor::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)>::operator()(librealsense::platform::frame_object, std::function<void()>) (__closure=0x21930e80, f=..., continuation=..., p=...) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/sensor.cpp:501
#11 0x00007f6fbe17dd1e in std::_Function_handler<void(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>), librealsense::uvc_sensor::open(const stream_profiles&)::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)> >::_M_invoke(const std::_Any_data &, <unknown type in /usr/lib/debug/.build-id/a8/9e6f6a5aa943744cf9053b49748a64d3a61ec1.debug, CU 0x7fed27, DIE 0x912e77>, <unknown type in /usr/lib/debug/.build-id/a8/9e6f6a5aa943744cf9053b49748a64d3a61ec1.debug, CU 0x7fed27, DIE 0x912e7c>, <unknown type in /usr/lib/debug/.build-id/a8/9e6f6a5aa943744cf9053b49748a64d3a61ec1.debug, CU 0x7fed27, DIE 0x912e81>) (__functor=..., __args#0=<optimized out>, __args#1=<optimized out>, __args#2=<optimized out>)
    at /usr/include/c++/5/functional:1871
#12 0x00007f6fbe1f5506 in std::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) const (__args#2=..., __args#1=..., __args#0=..., this=0x2f904b0) at /usr/include/c++/5/functional:2267
#13 librealsense::platform::v4l_uvc_device::poll (this=this@entry=0x2f90390) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/linux/backend-v4l2.cpp:862
#14 0x00007f6fbe1f5858 in librealsense::platform::v4l_uvc_device::capture_loop (this=0x2f90390) at /home/perc/workspace/RealsenseSDK_Gated_master__Debian_CreatePackage/src/src/linux/backend-v4l2.cpp:1238
#15 0x00007f6fa406f510 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#16 0x00007f6fa43426ba in start_thread (arg=0x7f6ec7fff700) at pthread_create.c:333
#17 0x00007f6fa37d041d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

thanks,
Omer.

omerbrandis, 2018-06-07:
note : 
i installed librealsense using ubuntu package installer ( apt-get) . i did not compile/build it intentionally/manually on my machine.

Omer.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
@omerbrandis
Can you let me know which Linux version you are using?

# Linux Distribution
**Intel® RealSense™ SDK 2.0** provides installation packages in `dpkg` format for Ubuntu 16 LTS*.
* The Realsense DKMS kernel drivers package (`librealsense2-dkms`) supports Ubuntu LTS kernels 4.4, 4.10 and 4.13.
**Note** Kernel 4.16 introduces some non-compatible modification to uvcvideo and media subsystem. We work to support it in future releases, but at this time the support is scoped up to kernel 4.15 and the affected users are requested to downgrade the kernel version.

> To build the project from source, or in case of Ubuntu 18 (Bionic Beaver) with kernel 4.15 please follow steps described here to manually install the kernel patches

omerbrandis, 2018-06-11:
Hello

I'm using ubuntu 16.04 LTS.

Thanks,
omer.

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
@omerbrandis
can you also let me know the kernel version? Thank you.

omerbrandis, 2018-06-12:
4.10.0-28-generic

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
@omerbrandis
Have you tried an clear machine to install librealsense? I've tried to install Ubuntu 16.04 on my 64 bits NUC device, then change the kernel to 4.10.0-28 and install librealsense2 follow the instruction, the realsense-viewer works well without problem as attachment. I didn't encouter the issue you posted.

carypai, 2018-06-19:
Attached image
[image]

omerbrandis, 2018-06-19:
Dear support,

I never said I get this error from realsense-viewer.
( I fact I can confirm that i don't get this error from the viewer)

I get this error when using librealsense from a program i've written .
( please see second sentance of the original post 
"i've written a program that uses two d415 cameras using librealsense 2.10.1"
)

Omer.

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
@omerbrandis
Sorry didn't see clear about your question, as you mentioned, this could be the conflict between your easylogging in the system to LibRS one. Since I don't have your code, so I can't reproduce the issue, could you try to build the LibRS from source code with easylogging disabled from CMAKE? please refer to PR attached. https://github.com/IntelRealSense/librealsense/pull/922

omerbrandis, 2018-06-20:
Hello,

I think it is very unlikely that the problem is due to a conflict with my installation/use of easylogger, since its an include only lib, i did not install it on my machine and the cureent version of my code does not use it explicitly ( it is only used by librealsense).

I realize its hard to reproduce the issue without my code, which i can't supply, 
But maybe you can suggest other additional diagnostic information that i may be able to provide that may help identify and debug this error ?

Thanks,
Omer.

omerbrandis, 2018-06-27:
i'm sorry , 

why was this issue closed ?
its still open , a fix has not been provided.

omer.

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
@omerbrandis
I am surprised as you, since it was not closed by me. Someone in my team accidentally close it, please re-open it, although right now I don't have other suggestion for you due to really hard to know where the problem is from only your log. But maybe someone in Github can provide you the suggestion or solution.

omerbrandis, 2018-06-28:
Thank you for reopening.
:-)

dorodnic, 2018-11-06:
Given the stack trace, may be related to #2674

eric-haibin-lin, 2020-04-19:
any update for the issue?

omerbrandis, 2020-04-20:
I have removed use of easylogger from my code , and thus aren't able to check this.
If you wish , you may close this ticket.
Omer.