realsenseai/librealsense #1801: LRS_LOG_LEVEL does not seem to work

omerbrandis, 2018-05-31:
hello, 

i am trying to control the logging level , 
followed instructions at https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs

but setting this env variable seems to have no effect on both my code, and on the standard applications.

here's an example 
> export LRS_LOG_LEVEL="FATAL"
> realsense-viewer
31/05 11:37:59,626 WARNING [140462828525312] (backend-v4l2.cpp:808) Empty frame has arrived.
 31/05 11:37:59,626 WARNING [140462828525312] (backend-v4l2.cpp:808) Empty frame has arrived.
 31/05 11:37:59,626 WARNING [140462828525312] (backend-v4l2.cpp:808) Empty frame has arrived.
 31/05 11:37:59,626 WARNING [140462828525312] (backend-v4l2.cpp:808) Empty frame has arrived.

(also tried export LRS_LOG_LEVEL=FATAL , to the same effect)

please advise.
Omer.

dorodnic, 2018-05-31:
I think [realsense-viewer.cpp:214](https://github.com/IntelRealSense/librealsense/blob/development/tools/realsense-viewer/realsense-viewer.cpp#L214) is overriding global settings. Did you try with other examples?

omerbrandis, 2018-06-07:
managed to change the logging level using rs2::log_to_console.
:-)

closed 2018-06-07 by omerbrandis