realsenseai/librealsense #1805: Have you provided python api for segmentation?

shinshiner, 2018-05-31:
I can get the segmentation results by running `rs-capture` command in terminal, but I have not found related api in python wrapper or docs. Therefore, I wonder if I can achieve the segmentation results from python. If not, how to use the results of `rs-capture` from python? Any help would be great appreciated!!

RealSense-Customer-Engineering, 2018-06-04:
[Realsense Customer Engineering Team Comment]
@shinshiner
There is no document specific to python, since Python wrapper is direct port of C++ API. There is also a way to check the API from python commend, please check the other issue link attached.
https://github.com/IntelRealSense/librealsense/issues/1710
https://github.com/IntelRealSense/librealsense/issues/1202

RealSense-Customer-Engineering, 2018-06-14:
[Realsense Customer Engineering Team Comment]
@shinshiner
Does the response answer you problem? can the issue be closed?

shinshiner, 2018-06-14:
Oh, very sorry for the delay, I think I have found some other algorithms written by Python to solve the issue. Many thanks for your answer !

closed 2018-06-14 by shinshiner