realsenseai/librealsense #1807: An issue about executing"./ZR300-live-test" with ZR300, not all passed.

melodybinbin, 2018-05-31:
**When I execute the command: ./ZR300-live-test, not all cases pass. And the terminal shows as following:**

`yuanlibin@yuanlibin:~/librealsense-1.12.1/build/unit-tests$ ./ZR300-live-test 
Camera type Intel RealSense ZR300

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ZR300-live-test is a Catch v1.2.1 host application.
Run with -? for options

-------------------------------------------------------------------------------
DS-device verify standard UVC Controls set/get
-------------------------------------------------------------------------------
/home/yuanlibin/librealsense-1.12.1/unit-tests/unit-tests-live-ds-common.cpp:692
...............................................................................

/home/yuanlibin/librealsense-1.12.1/unit-tests/unit-tests-live-ds-common.cpp:763: FAILED:
  REQUIRE( modified_values[i]==verification_values[i] )
with expansion:
  167.0 == 166.0

Camera type Intel RealSense ZR300
Camera type Intel RealSense ZR300
===============================================================================
test cases:     66 |     65 passed | 1 failed
assertions: 172375 | 172374 passed | 1 failed

yuanlibin@yuanlibin:~/librealsense-1.12.1/build/unit-tests$`

**It has confused me a few days. Looking forward to your response. Many thanks in advance!**

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
Product has end-of-lifed; no feature fixes or product enhancements planned

closed 2018-06-27 by RealSense-Customer-Engineering

melodybinbin, 2018-08-01:
@RealSense-Customer-Engineering 
OK, Thank you all the same!