realsenseai/librealsense #1810: Multi-camera synchronization for rosbag files

MatsumotoJ, 2018-05-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 10  |                                 
| Platform                                 | PC  |
| SDK Version                            |  2.10.2  | 

Hello.

I am trying to record a 360 deg 3D video using multiple D400 cameras. 
I want to use rosbag files (one file for each camera) for saving the data because of its low computational load.

Here is the code that I tried for the recording:
```
rs2::context rs_ctx;
const std::string platform_camera_name = "Platform Camera";
i = 0;
for (auto&& dev : rs_ctx.query_devices()) 
{
    if (dev.get_info(RS2_CAMERA_INFO_NAME) == platform_camera_name) continue;

    rs2::pipeline pipe;
    rs2::config cfg;

    cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
    cfg.enable_stream(RS2_STREAM_COLOR, 848, 480, RS2_FORMAT_BGR8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 30);
    cfg.enable_record_to_file("camera." + std::to_string(i+1) + ".bag");

    rs2::pipeline_profile prof = pipe.start(cfg);

    i++;
}
```
However the recorded files were not synchronized: some cameras start earlier than the others; there also could be frame drops at the different timings for different cameras. 

I have following two idea for the synchronization:
- synchronizing the recorded files in offline by using frame timestamp data and `rs2::playback::seek`.
- synchronizing recording/playback pipelines by using `rs2::syncer`.

But I could not find enough documentations for either function, so I have not successfully implemented the ideas yet. 

Could you kindly provide an example of a code for synchronous recording/playback rosbag of multi cameras?

dorodnic, 2018-06-03:
Hi @MatsumotoJ 
Your analysis is 100% correct. We are planning to address these points in the future.
In addition, we have recently added [rs-convert](https://github.com/IntelRealSense/librealsense/tree/development/tools/convert) util to extract frames from BAG files as a post-processing step.

MatsumotoJ, 2018-06-03:
Hi dorodnic,

Thank you very much for the info. 
The code of rs-convert looks a good example to start implementing the first idea. 
I would share the implementation here, after I success. 

Thank you again.

MatsumotoJ, 2018-06-04:
Hi again,

Here is the implementation:
https://gist.github.com/MatsumotoJ/e23a80f540b58db12ee92d8a96de1745
(There could be &plusmn; 1.0/fps lag in the present implementation; more precise synchronization for &plusmn; 0.5/fps lag, would be possible)

Thank you very much for your help.

closed 2018-06-05 by MatsumotoJ

hflsdupont, 2018-07-18:
Hi, @MatsumotoJ , 

How do you make sure that multiple cameras have the same clock? Since the timestamp is based on camera's own hardware clock, I find that timestamps cannot be directly compared between two cameras.

MatsumotoJ, 2018-07-19:
Hi @hflsdupont,

I have not checked extensively but so far the code above works for my environment (Firmware difference?). 
I checked it by capturing a moving object in front of the cameras and then reconstructing a full 3D video with the synchronized frames. 

Do you have any idea, @dorodnic ?