realsenseai/librealsense #1811: Related to Calibration Procedure for Large Outdoor Depth Distance

Parikshit01, 2018-06-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version | Win (10) |
| Platform                                 | PC  |
| SDK Version                            | 2.11.0  | 

### Issue Description
I am using D435 for drone's vision application in outdoor environment where the depth distance need to be found is around 30m. With the default dynamic calibrator given with SDK am only able to get a max of 10m distance within which range the object can be easily segmented. Changing the 40 differnent parameters we can manually change calibration settings of the camera for longer range as per lighting conditions available. But am not able to do dynamically calibrate specific for my application. In documentation it's written upto 65m to 325m its possible to report range for D400 series. Also in some youtube video i've found they can detect upto 80m. Please guide me. 

[image]
[image]

agrunnet, 2018-06-01:
The range will depend on the camera model and on the error that you can tolerate. In the same presentation there is a graph of the error vs distance. 
The D435 is not intended for long range. The D415 has 2.5x less error at same range. Other models exits that can even better performance at long range but are for development purposes only at this point.

Changing the 40 parameters do not change the calibration but it mainly affects how depth is reported. Specifically for you if you want data beyond 65m you will need to adjust the advanced parameter called “depth units” to 5000 for example and then the app will report our values for depth longer. Did you try adjusting the depth units?

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
hi @Parikshit01,

do you need more support on this topic?

Parikshit01, 2018-06-21:
Hi, My problem has still remained unsolved. Anyways I've one query. if perfectly calibration is done, with D435 in indoor environment & in outdoor environment up to what distance it's possible to perfectly measure the depth and segment it from the background.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
hi @Parikshit01,

would like to know if you tried the depth unit as agrunnet suggest?
D435 to measure depth is mentioned in chapt 6.5, in the datasheet, it's depend on how precisely you want to measure the depth, at 30meter distance, it not common usage for D435, I don't have report at this distance.
and for segment from background, this is depend on the algorithm to do the segmentation.

closed 2018-07-16 by RealSense-Customer-Engineering