realsenseai/librealsense #1812: How to transform pixels coordinate in the color frame to the coordinates in the depth frame (I have 2D rectangle in a color frame and I wanna transform its coordinates to the corresponding depth frame)? Is the align example what I need?

AkhmedAnam, 2018-06-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense-Customer-Engineering, 2018-06-04:
[Realsense Customer Engineering Team Comment]
@AkhmedAna
Yes, the rs-align.cpp is what you need, you can create the align instance with Depth stream as parameter then when you call the process of align instance, it will align the color image to the depth image.

RealSense-Customer-Engineering, 2018-06-14:
[Realsense Customer Engineering Team Comment]
@AkhmedAna
The align example should solve your question, close the issue first, if you still have problem, please reopen it or create new issue.

closed 2018-06-14 by RealSense-Customer-Engineering