realsenseai/librealsense #1813: realsenviewer fails to show stream in few seconds. plug out plug in camera will work again for few seconds. same error with code examples

contactprad, 2018-06-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } D415| 
| Firmware Version                   | (Open RealSense Viewer --> Click info)05.09.11.00 | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  |  Ubuntu 16
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)     4.4.0.2-upboard                                    | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }      2.11                    | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Realsense viewer will show the stream for some seconds and then throws the below error:

01/06 17:25:49,025 ERROR [140650463545088] (backend-v4l2.cpp:1243) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 01/06 17:25:49,031 ERROR [140650381047552] (types.h:179) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 01/06 17:25:49,032 ERROR [140650381047552] (backend-v4l2.cpp:1243) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 01/06 17:25:49,058 ERROR [140650480330496] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 01/06 17:25:49,058 ERROR [140650480330496] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 01/06 17:25:49,068 ERROR [140650372654848] (backend-v4l2.cpp:265) xioctl(VIDIOC_QBUF) failed when requesting new frame! Last-error: No such device
 01/06 17:25:49,068 ERROR [140650372654848] (types.h:179) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 01/06 17:25:49,068 ERROR [140650372654848] (backend-v4l2.cpp:1243) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
 01/06 17:25:50,059 ERROR [140650480330496] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 01/06 17:25:50,059 ERROR [140650480330496] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 01/06 17:25:50,916 WARNING [140650534688512] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 01/06 17:25:51,059 ERROR [140650480330496] (types.h:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 01/06 17:25:51,060 ERROR [140650480330496] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 01/06 17:25:51,162 WARNING [140650806491712] (backend-v4l2.cpp:1156) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 01/06 17:25:51,166 WARNING [140650806491712] (sensor.cpp:330) Unregistered Media formats : [ RW16 ]; Supported: [ ]
 01/06 17:25:51,275 ERROR [140650806491712] (sensor.cpp:253) An error has occurred while stop_streaming()!
 01/06 17:25:51,281 ERROR [140650806491712] (types.h:179) xioctl(VIDIOC_STREAMOFF) failed Last Error: No such device
 01/06 17:25:51,282 ERROR [140650806491712] (sensor.cpp:253) An error has occurred while stop_streaming()!
 01/06 17:25:51,282 WARNING [140650806491712] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x2df5820 stopped
 01/06 17:25:51,282 WARNING [140650806491712] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x2df5820 stopped
 01/06 17:25:51,287 WARNING [140650806491712] (archive.cpp:326) All frames from stream 0x0x2df5820 are now released by the user
 01/06 17:25:51,954 ERROR [140650349577984] (types.h:179) Cannot open '/dev/video3 Last Error: No such file or directory

Example code and other code throws the same error.

dorodnic, 2018-06-03:
If you are experiencing sporadic disconnects, we recommend to check the USB3 cable and retry the experiment with **powered** USB3 hub.

RealSense-Customer-Engineering, 2018-06-14:
[Realsense Customer Engineering Team Comment]
@contactprad
Have you changed the powered USB3 hub and cable? does the issue still exist?

closed 2018-06-22 by RealSense-Customer-Engineering

6467121626, 2018-06-30:
we also have the same issue, the device can be found with "lsusb", but no data received. plug in and out solve this.

contactprad, 2018-11-12:
We have cable which came as part of kit and it works fine on desktop but not on Up-board.

jungladicitta, 2019-08-08:
> We have cable which came as part of kit and it works fine on desktop but not on Up-board.

how did you solved this issue?