realsenseai/librealsense #1814: Android Studio building Error

fotsing365, 2018-06-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {ZR300  } | 
| Firmware Version                   | | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.14.13)                                      | 
| Platform                                 |  |
| SDK Version                            |  { legacy / 2.11.1 }                          | 

<While following the instructions [here,](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md) getting the following runtime error   
org.gradle.internal.UncheckedException: Build command failed.
Error while executing process /home/cedriq/Android/Sdk/cmake/3.6.4111459/bin/cmake with arguments {--build /home/cedriq/bin/Android/Project/RealSense/app/.externalNativeBuild/cmake/debug/arm64-v8a --target realsense2}
[1/106] Building C object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o
[2/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/core.c.o
[3/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/descriptor.c.o
[4/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/hotplug.c.o
[5/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/io.c.o
[6/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/strerror.c.o
[7/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/sync.c.o
[8/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/os/linux_usbfs.c.o
[9/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/os/poll_posix.c.o
[10/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/os/threads_posix.c.o
[11/106] Building C object src/main/cpp/librealsense/third-party/libusb/CMakeFiles/usb.dir/libusb/os/linux_netlink.c.o
[12/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/console_bridge/src/console.cpp.o
/home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/core.c:2328:10: warning: enumeration value 'LIBUSB_LOG_LEVEL_NONE' not handled in switch [-Wswitch]
        switch (level) {
                ^
1 warning generated.
[13/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/debug.cpp.o
[14/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/header.cpp.o
[15/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag.cpp.o
In file included from /home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/os/linux_netlink.c:45:
/home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/os/linux_usbfs.h:102:46: warning: zero size arrays are an extension [-Wzero-length-array]
        struct usbfs_iso_packet_desc iso_frame_desc[0];
                                                    ^
/home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/os/linux_usbfs.h:140:20: warning: zero size arrays are an extension [-Wzero-length-array]
        unsigned char eps[0];
                          ^
2 warnings generated.
[16/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag_player.cpp.o
In file included from /home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/os/linux_usbfs.c:43:
/home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/os/linux_usbfs.h:102:46: warning: zero size arrays are an extension [-Wzero-length-array]
        struct usbfs_iso_packet_desc iso_frame_desc[0];
                                                    ^
/home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/libusb/libusb/os/linux_usbfs.h:140:20: warning: zero size arrays are an extension [-Wzero-length-array]
        unsigned char eps[0];
                          ^
2 warnings generated.
[17/106] Linking C static library src/main/cpp/librealsense/third-party/libusb/libusb.a
[18/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/buffer.cpp.o
[19/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/chunked_file.cpp.o
[20/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/lz4_stream.cpp.o
[21/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/message_instance.cpp.o
In file included from /home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/realsense-file/rosbag/cpp_common/src/header.cpp:37:
../../../../src/main/cpp/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:79:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_ERROR, fmt, ##__VA_ARGS__)
                                                                                         ^
../../../../src/main/cpp/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:79:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
2 warnings generated.
[22/106] Building CXX object src/main/cpp/librealsense/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/query.cpp.o
In file included from /home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/lz4_stream.cpp:39:
../../../../src/main/cpp/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:79:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_ERROR, fmt, ##__VA_ARGS__)
                                                                                         ^
...............
In file included from /home/cedriq/bin/Android/Project/RealSense/app/src/main/cpp/librealsense/src/device.cpp:5:
In file included from ../../../../src/main/cpp/librealsense/src/device.h:12:
In file included from ../../../../src/main/cpp/librealsense/src/hw-monitor.h:6:
../../../../src/main/cpp/librealsense/src/sensor.h:151:86: warning: multi-character character constant [-Wfour-char-constants]
        const std::map<rs2_stream, uint32_t> stream_and_fourcc = {{RS2_STREAM_GYRO,  'GYRO'},
[... 2005 lines cut ...]
    inline std::ios_base& hexfloat(std::ios_base& str)
                          ^
../../../../src/main/cpp/librealsense/src/types.h:395:5: note: candidate function not viable: no known conversion from 'librealsense::to_string' to 'std::ostream &' (aka 'basic_ostream<char> &') for 1st argument
    RS2_ENUM_HELPERS(rs2_stream, STREAM)
    ^
../../../../src/main/cpp/librealsense/src/types.h:385:31: note: expanded from macro 'RS2_ENUM_HELPERS'
        inline std::ostream & operator << (std::ostream & out, TYPE value) { if(is_valid(value)) return out << get_string(value); else return out << (int)value; } \
                              ^
../../../../src/main/cpp/librealsense/src/types.h:396:5: note: candidate function not viable: no known conversion from 'librealsense::to_string' to 'std::ostream &' (aka 'basic_ostream<char> &') for 1st argument
    RS2_ENUM_HELPERS(rs2_format, FORMAT)
    ^
[... cut ...]
                              ^
../../../../src/main/cpp/librealsense/src/types.h:398:5: note: candidate function not viable: no known conversion from 'librealsense::to_string' to 'std::ostream &' (aka 'basic_ostream<char> &') for 1st argument
    RS2_ENUM_HELPERS(rs2_option, OPTION)
    ^
../../../../src/main/cpp/librealsense/src/types.h:385:31: note: expanded from macro 'RS2_ENUM_HELPERS'
        inline std::ostream & operator << (std::ostream & out, TYPE value) { if(is_valid(value)) return out << get_strin

fotsing365, 2018-06-04:
I changed my NDK TO 16Bb and this fix the problem. Hope somebody can explain why?

fotsing365, 2018-06-04:
I cahnged my NDK from 17 to 16b and this fix the problem. Hope somebody can explain me why

dorodnic, 2018-06-05:
It is most likely the same issue as #1776, we will try to add switch to support both

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
@fotsing365 Is the issue resolved?

fotsing365, 2018-07-10:
Hi  Excuse me .I have had fixed this error just by adding     **#include <atomic>**  in the [native-lib.cpp](https://github.com/IntelRealSense/librealsense/blob/v2.12.0/doc/android/native-lib.cpp_) it seems like you have forgotten.