realsenseai/librealsense #1815: Advanced settings: Color correction values persist after loading a new preset / JSON.

xtrawurst, 2018-06-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 2.11.1  | 
| Operating System & Version | Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.11.1 | 

To reproduce:

1. Connect RS camera, Start RS viewer. Choose the "Default" Preset. Then open the stereo module on the left, select "RGB8" as the IR1 format and turn on the stereo module. Choosing the "Default" preset will reset the color correction.

2. Turn off the stereo module and open the attached "noir.json" settings file as a preset or choose the "Remove IR pattern" preset. Turn on the stereo module again.

3. Turn off the stereo module and open the attached "empty.json" settings file or choose another preset (except for the **Default** preset). Turn on the stereo module.

4. The colors of the left imager are still modified and the IR pattern is still removed. The expected behavior is that settings of a previously loaded preset / JSON have no effect on the current preset / JSON, hence the expectation is that the IR pattern is visible again.

agrunnet, 2018-06-01:
Please try latest FW and let us know if you still see this. This was a bug that should be have been fixed.

xtrawurst, 2018-06-01:
The camera is running 05.09.11.00 (according to the RS Viewer). Is there a newer one available?

agrunnet, 2018-06-01:
Yes that is latest. I think I misunderstood the question then. 
It sounds like you are saying that if you load a new preset that has NO COLOR CORRECTION, then you want to revert to something else? The behavior now should be that a JSON file can contain many preset values but not all values. So if you omit any then we leave those values as the previous ones. Not all presets include values of the COLOR CORRECTION.  The reason for this is that we want to allow users to use a specific Depth Preset and then switch between different color presets, for example, without affecting depth. 
Did I get the question correct this time?

xtrawurst, 2018-06-01:
> Did I get the question correct this time?

Yes. Our expectation was that every omitted setting in a preset / JSON would result in a default value being set. It's a bit counter-intuitive that settings made in previous presets / JSONs can have an effect on the current state.

I guess the workaround to get the desired behaviour would be to always load up a default JSON / preset first and then to load up any specific settings **on top**. Is there an API call to load up the default preset directly without going through JSON?

The best solution would be if we could specify in the JSON whether it should set the omitted values to default or whether it should keep them unchanged and use whatever was set before.

xtrawurst, 2018-06-04:
@dorodnic 

Ok. Maybe a workaround would be to release a JSON file that sets **every** value to its default? Could you do that?
I have tried selecting the "Default" preset in the Viewer and saving that to a JSON but it seemed that not every value ended up in the JSON file.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

I thought if you save preset after hardware_reset will save all default values into file, then you can use it as "default" preset for you, may I know which value is missed if you do so?
thanks.

xtrawurst, 2018-06-08:
@RealSense-Customer-Engineering 
I think the last time I tried this it did not save all the values. I will try this again.
Thanks.

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
hi xtrawurst,

thanks, please let me know if we miss any

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

would like to know if your test is good?

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @@xtrawurst,

i think the latest release of libRealsense fixed issue with load/save setting, please let me know if if still having this issue.

xtrawurst, 2018-07-02:
@RealSense-Customer-Engineering I will try and let you know.

xtrawurst, 2018-07-02:
I was just trying to reproduce the issue using the latest RS Viewer and FW. The result is issue #1988.

The issue in this thread reproduces as described when I use a D415 with FW 5.9.11 and the latest RS Viewer.

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

i tried to start realsense viewer with D415 connected and then save setting to file as default.json.
follow your test steps, it works good, at least i can tell remove IR on/off by loading your noir.json and my default.json.
please let me know if this work for you or any parameter missed, thanks.

xtrawurst, 2018-07-09:
@RealSense-Customer-Engineering 

The issue should still be reproducible with the latest RS viewer / FW. However I understand this is the desired behavior.

I tested saving the Default setting to a JSON and that seems to work fine now, so it's an acceptable workaround.

closed 2018-07-09 by xtrawurst