realsenseai/librealsense #1816: Segfault when using a threads in combination with gdb

kevindehecker, 2018-06-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400                                  | 
| Firmware Version                   | 05.09.11.00                        | 
| Operating System & Version |   Ubuntu 16                         | 
| Kernel Version (Linux Only)    |  4.13.0-43-generic             | 
| Platform                                 | PC                                       |
| SDK Version                            |  v2.11.1                             | 

### Issue Description
When I use the RealSense camera from a new thread and break in the main thread with gdb (I use the QtCreator IDE for that), wait for 10 seconds, and continue, it segfaults when I continue.  

I have made skeleton code to demonstrate the issue: https://github.com/kevindehecker/rs-test
The project simply shows the video obtained from the left depth camera in an opencv imshow window. There are two modes, the threaded and none threaded mode.

Steps to reproduce:
1. build and start the project (it shows the video window) from https://github.com/kevindehecker/rs-test
2. set a breakpoint / break somewhere in the while loop
3. wait 10 seconds
4. continue

Step 4  results in a segfault from the realsense api. If the api is used in non-threaded code, the segfault disappears. This can be done by  setting the `#if 0`  to `1` on line 7 in main.cpp.

How to proceed?

dorodnic, 2018-06-03:
Hi @kevindehecker 
If the error disappears in non-threaded code, what prevents you from using it? 
If I understand correctly, the exception is thrown on the new thread you create and you can add appropriate exception handling code there to avoid the crash.

kevindehecker, 2018-06-03:
Right, catch the error. I can't believe I didn't think of this my self :S 
(Initially I was under the impression this thing was segfaulting, but that turned out to be caused by something else entirely, probably didn't think much further)
Oh well, I still think it's a bit weird this error is thrown only in the threaded case, but if the devs are ok with that so am I... (feel free to close this issue in that case)

> If the error disappears in non-threaded code, what prevents you from using it?

The reason I want the RealSense code in a new thread, is that the non-essential visualizations are being drawn from the main thread. In particular OpenCV imshow does not like to be in anything else than the main thread. In order to guarantee full fps processing of the images, the essential realsense camera algorithms are processed in a separate thread.

dorodnic, 2018-06-03:
Sorry, I just wanted to get some context, your approach certainly makes sense. 
You can catch the error from `wait_for_frames` or use `poll_for_frames` with a custom delay. Alternatively, in some SDK examples, we keep the frame grabbing loop in the main thread, but then send frames for processing to another thread using `frame_queue` concurrent queue primitive.

kevindehecker, 2018-06-03:
Yes, I'm solving it now by catching. I was just wondering why this exception is only being thrown when `wait_for_frames`  is in another thread! But at this point it's only a curiosity, as the catching solution works fine for me.