realsenseai/librealsense #1817: No UVC backend selected. Please #define exactly one of RS2_USE_LIBUVC_BACKEND, RS2_USE_WMF_BACKEND, or RS2_USE_V4L2_BACKEND

xufeifeiWHU, 2018-06-02:
when I try to install librealsense SDK 2 with python,
the ditctory is  ../wrapper/python
I have followed the guides  which says

    mkdir build
    cd build
    cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true

    make -j4                                 <----error occurs
    sudo make install

error:No UVC backend selected. Please #define exactly one of RS2_USE_LIBUVC_BACKEND, RS2_USE_WMF_BACKEND, or RS2_USE_V4L2_BACKEND

After I have checked the ../librealsense/src/backend.cpp file,
It seems if I install libusb-1.0 or libuvc may be useful , but it's failed to solve the error.
Thank you for your time.

xufeifeiWHU, 2018-06-03:
OK  I have solve the problem:
the problem occurs when I run “cmake -D。。。。” under the path which is "wrapper/python/build",
but  it's incorrect at all,
the guides says like this：
“Run the top level CMake command with the following additional flag -DBUILD_PYTHON_BINDINGS=bool:true:”
SO，I should run cmake under the /librealsense/build and pass the flag -DBUILD_PYTHON_BINDINGS=bool:true
rather than under the /librealsense/wrapper/python/build.
Check your path clearly.

dorodnic, 2018-06-03:
Hi @xufeifeiWHU 
Sorry for the confusion but glad you figured it out. Feel free to suggest modifications to the guide via a pull-request.

RealSense-Customer-Engineering, 2018-06-04:
[Realsense Customer Engineering Team Comment]
Hi @xufeifeiWHU ,
Glad you figured out your problem.  We will close this request and please open a new one if you have any other issues.

closed 2018-06-04 by RealSense-Customer-Engineering

CChape, 2018-08-29:
Hi, I have a problem with the installation of the pyrealsense2 package for python. 
I make the librealsense construction with the appopriate flag, and everything work, but when I want to use it in python:
import pyrealsense2
or
import pyrealsense 
I got: "no module named .."
Do you have any suggestion?
I'm on the Jetson TX2 and the camera is the D435 RealsSense.

Thanks in advance 
C. Chappey

freemanlo, 2018-08-29:
You can review the instructions:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python