realsenseai/librealsense #1818: SR300 - The requested pixel format 'INZI' is not natively supported by the Linux kernel

msieb1, 2018-06-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300} | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Linux (Ubuntu 14)  | 
| Kernel Version (Linux Only)    |  4.4.0-112-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            | 2.11.0                  | 

### Issue Description
The SR300 does not supply a depth stream. When opening realsense-viewer, the error appears as mentioned in the title.

It saws to upgrade the kernel to 4.12 or later, but the kernel is already at version 4.4.0.

dorodnic, 2018-06-03:
Hi @msieb1 
Kernel 4.12 is about 1 year old and 4.4 is 3 years old. 
The message is accurate - you need to either apply the Kernel patch as described in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) or upgrade to newer kernel.

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
hi @msieb1,

do you need more support on this topic?

msieb1, 2018-06-21:
This is solved. Thank you very much.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
thanks for confirm, let me close this ticket here and feel free to create new ticket if any question,

closed 2018-06-22 by RealSense-Customer-Engineering