realsenseai/librealsense #1819: Unable to get raw Left & Right IR streams from D415

arunrajagopalan, 2018-06-03:
* Probably related to #1140 and #1707, but opening new issue since I am using D415 and not using OpenCV

* Tried various combinations of resolution, format, fps, stream indexes, not specifying stream indexes, enabling depth stream like suggested in #1140. But I cant get both raw streams. I can get one raw stream and the depth

* I included the rs-sensor-control output on supported streams at the end
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                       | 
| Firmware Version                   | 05.09.11.00                                             | 
| Operating System & Version | Win 10                                                    | 
| Platform                                 | PC                                                           |
| SDK Version                           |  2                                                            | 

### Issue Description
Unable to get left & right raw IR from D415

I am getting the exception 
RealSense error calling rs2_pipeline_start_with_config(pipe:0000021B41720E90, config:0000021AFCE11240):
 **Failed to resolve request. No device found that satisfies all requirements**

    	cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
    	cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
    
    	//cfg.enable_stream(RS2_STREAM_INFRARED, 0, 640, 480, RS2_FORMAT_UYVY); // << works by itself, gets index 0
    	//cfg.enable_stream(RS2_STREAM_INFRARED, -1, 640, 480, RS2_FORMAT_Y8); // << works by itself, gets index 1
    	//cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8); // << works by itself, gets index 1
    	//cfg.enable_stream(RS2_STREAM_INFRARED, 2);
    	//cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
    	//cfg.enable_stream(RS2_STREAM_DEPTH);
    	// Declare RealSense pipeline, encapsulating the actual device and sensors
    	rs2::pipeline pipe;
    	// Start streaming with default recommended configuration
    	pipe.start(cfg);
    
    

0  : Infared #0 (Video Stream: UYVY 1280x720@ 6Hz)
1  : Infrared #1 (Video Stream: Y8 1280x720@ 6Hz)
2  : Infrared #0 (Video Stream: BGRA8 1280x720@ 6Hz)
3  : Infrared #0 (Video Stream: RGBA8 1280x720@ 6Hz)
4  : Infrared #0 (Video Stream: BGR8 1280x720@ 6Hz)
5  : Infrared #0 (Video Stream: RGB8 1280x720@ 6Hz)
6  : Infrared #0 (Video Stream: UYVY 1280x720@ 4Hz)
7  : Infrared #1 (Video Stream: Y8 1280x720@ 4Hz)
8  : Infrared #0 (Video Stream: BGRA8 1280x720@ 4Hz)
9  : Infrared #0 (Video Stream: RGBA8 1280x720@ 4Hz)
10 : Infrared #0 (Video Stream: BGR8 1280x720@ 4Hz)
11 : Infrared #0 (Video Stream: RGB8 1280x720@ 4Hz)
12 : Infrared #0 (Video Stream: UYVY 640x480@ 30Hz)
13 : Infrared #1 (Video Stream: Y8 640x480@ 30Hz)
14 : Infrared #0 (Video Stream: BGRA8 640x480@ 30Hz)
15 : Infrared #0 (Video Stream: RGBA8 640x480@ 30Hz)
16 : Infrared #0 (Video Stream: BGR8 640x480@ 30Hz)
17 : Infrared #0 (Video Stream: RGB8 640x480@ 30Hz)
18 : Infrared #0 (Video Stream: UYVY 640x480@ 15Hz)
19 : Infrared #1 (Video Stream: Y8 640x480@ 15Hz)
20 : Infrared #0 (Video Stream: BGRA8 640x480@ 15Hz)
21 : Infrared #0 (Video Stream: RGBA8 640x480@ 15Hz)
22 : Infrared #0 (Video Stream: BGR8 640x480@ 15Hz)
23 : Infrared #0 (Video Stream: RGB8 640x480@ 15Hz)
24 : Infrared #0 (Video Stream: UYVY 640x480@ 6Hz)
25 : Infrared #1 (Video Stream: Y8 640x480@ 6Hz)
26 : Infrared #0 (Video Stream: BGRA8 640x480@ 6Hz)
27 : Infrared #0 (Video Stream: RGBA8 640x480@ 6Hz)
28 : Infrared #0 (Video Stream: BGR8 640x480@ 6Hz)
29 : Infrared #0 (Video Stream: RGB8 640x480@ 6Hz)
30 : Infrared #0 (Video Stream: UYVY 640x480@ 4Hz)
31 : Infrared #1 (Video Stream: Y8 640x480@ 4Hz)
32 : Infrared #0 (Video Stream: BGRA8 640x480@ 4Hz)
33 : Infrared #0 (Video Stream: RGBA8 640x480@ 4Hz)
34 : Infrared #0 (Video Stream: BGR8 640x480@ 4Hz)
35 : Infrared #0 (Video Stream: RGB8 640x480@ 4Hz)
36 : Infrared #0 (Video Stream: UYVY 480x270@ 60Hz)
37 : Infrared #1 (Video Stream: Y8 480x270@ 60Hz)
38 : Infrared #0 (Video Stream: BGRA8 480x270@ 60Hz)
39 : Infrared #0 (Video Stream: RGBA8 480x270@ 60Hz)
40 : Infrared #0 (Video Stream: BGR8 480x270@ 60Hz)
41 : Infrared #0 (Video Stream: RGB8 480x270@ 60Hz)
42 : Infrared #0 (Video Stream: UYVY 480x270@ 30Hz)
43 : Infrared #1 (Video Stream: Y8 480x270@ 30Hz)
44 : Infrared #0 (Video Stream: BGRA8 480x270@ 30Hz)
45 : Infrared #0 (Video Stream: RGBA8 480x270@ 30Hz)
46 : Infrared #0 (Video Stream: BGR8 480x270@ 30Hz)
47 : Infrared #0 (Video Stream: RGB8 480x270@ 30Hz)
48 : Infrared #0 (Video Stream: UYVY 480x270@ 15Hz)
49 : Infrared #1 (Video Stream: Y8 480x270@ 15Hz)
50 : Infrared #0 (Video Stream: BGRA8 480x270@ 15Hz)
51 : Infrared #0 (Video Stream: RGBA8 480x270@ 15Hz)
52 : Infrared #0 (Video Stream: BGR8 480x270@ 15Hz)
53 : Infrared #0 (Video Stream: RGB8 480x270@ 15Hz)
54 : Infrared #0 (Video Stream: UYVY 480x270@ 6Hz)
55 : Infrared #1 (Video Stream: Y8 480x270@ 6Hz)
56 : Infrared #0 (Video Stream: BGRA8 480x270@ 6Hz)
57 : Infrared #0 (Video Stream: RGBA8 480x270@ 6Hz)
58 : Infrared #0 (Video Stream: BGR8 480x270@ 6Hz)
59 : Infrared #0 (Video Stream: RGB8 480x270@ 6Hz)
60 : Infrared #0 (Video Stream: UYVY 480x270@ 4Hz)
61 : Infrared #1 (Video Stream: Y8 480x270@ 4Hz)
62 : Infrared #0 (Video Stream: BGRA8 480x270@ 4Hz)
63 : Infrared #0 (Video Stream: RGBA8 480x270@ 4Hz)
64 : Infrared #0 (Video Stream: BGR8 480x270@ 4Hz)
65 : Infrared #0 (Video Stream: RGB8 480x270@ 4Hz)
66 : Depth #0 (Video Stream: Z16 1280x720@ 6Hz)
67 : Depth #0 (Video Stream: Z16 1280x720@ 4Hz)
68 : Depth #0 (Video Stream: Z16 640x480@ 30Hz)
69 : Depth #0 (Video Stream: Z16 640x480@ 15Hz)
70 : Depth #0 (Video Stream: Z16 640x480@ 6Hz)
71 : Depth #0 (Video Stream: Z16 640x480@ 4Hz)
72 : Depth #0 (Video Stream: Z16 480x270@ 60Hz)
73 : Depth #0 (Video Stream: Z16 480x270@ 30Hz)
74 : Depth #0 (Video Stream: Z16 480x270@ 15Hz)
75 : Depth #0 (Video Stream: Z16 480x270@ 6Hz)
76 : Depth #0 (Video Stream: Z16 480x270@ 4Hz)`

dorodnic, 2018-06-03:
Hi @arunrajagopalan 
What about realsense-viewer? Can you find the desired configuration in the viewer? 
If I recall correctly, the D400 come out of the box with Right IR stream disabled but you can unlock it by activating advanced mode (the simplest way is by activating one of the visual presets from the viewer).

dorodnic, 2018-06-04:
Ok, so if you can modify presets and use the advanced mode params, the camera is already in advanced mode, that's good. 

What version of the SDK you use?
We had a bug prior to 2.11.1 (@ev-mp, can you confirm please) that affected specific Windows 10 versions (RS3 build) and may be related to what you experience.

ev-mp, 2018-06-04:
Hello @arunrajagopalan ,
The log and the description indicate that D415 is recognized as USB2 (and not USB3) device. In USB2 mode the device is designed to expose only a subset of streaming profiles according to USB bandwidth budget.
Running Realsense-Viewer will present the USB connection type. Make sure it displays USB3+ mode, otherwise check the cable/USB connectors/port to fix it.

arunrajagopalan, 2018-06-04:
@ev-mp Realsense-Viewer is showing USB 3.1 when connected to one port and USB 3.0 when connected to another port

arunrajagopalan, 2018-06-04:
@dorodnic SDK version is 2.11.1

ev-mp, 2018-06-04:
Hi @arunrajagopalan , what is the windows build you're using (type "winver").
The port that reports 3.1 is configured correctly. You should be able to stream Left+Right IR/HD in 30fps.

With regards to 3.0 - it is not common that a PC has multiple USB3 controllers, are you using USB3 PCI-E extension?
Additionally, can you post output of `rs-enumerate-devices` when connecting the camera to both ports ?

arunrajagopalan, 2018-06-05:
@ev-mp  winver is Version 1803, OS Build 17134.48
Attachment has enumerate devices output

PS: my responses may be delayed for a couple days. I appreciate very much your help!

[rs-enumerate-devices.txt](https://github.com/IntelRealSense/librealsense/files/2072242/rs-enumerate-devices.txt)

RealSense-Customer-Engineering, 2018-06-06:
[Realsense Customer Engineering Team Comment]
hi @arunrajagopalan,

I've confirmed D415 with Realsense viewer, I can do IR stream from Left and Right IR camera at the same time under 1280x720x30fps, from your rs-enumerate-devices also say this is supported.
since you mentioned your realsense viewer have only single IR stream, can you please confirm that D415 us showing USB3.1 and you checked Infra 1 and Infra 2 under Stereo Module when stream is off?
if so can you please find another PC and see if the issue can reproduce on the other PC?
would like to identify if this issue is PC depended.
thanks.

arunrajagopalan, 2018-06-06:
Folks, I got a different PC . I am able to get the left and right IR from the d415 with 1280x720 x 30 fps on a USB 3.1 on that PC. The posted code also works and gets both streams

On the first computer … an ideapad, the realsense viewer shows both IR left & right on the 3.1 port,.

 However the code to get the streams fails. Not sure why. So @RealSense-Customer-Engineering I agree its PC dependent. But since the viewer sees both cameras but the posted code doesn't, I would think somethings wrong with my code too

arunrajagopalan, 2018-06-07:
Fols, I can get both streams on the New PC now, even in the code. So this is definitely PC dependent.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @arunrajagopalan,

thanks for confirm, so maybe we can close this ticket and please feel free to create new ticket if any follow up question?

arunrajagopalan, 2018-06-10:
I think we can close it, but it would be good to know what about the PC makes it not work

closed 2018-06-10 by arunrajagopalan

amaanda, 2019-02-21:
I am having the frame drop problem on a notebook. Any advances in knowing what it is about the PC that may cause problems?

ShravyaB, 2020-08-04:
I am facing a weird issue. Using code or realsense viewer, I am not able to simultaneously get both infrared images. (2 images show up but they don't have any offset/disparity between them)
But if I stream only left and right independently, then I get the different images (disparity is present)
Any idea what is causing this?