realsenseai/librealsense #1821: Segmentation Fault after several seconds

KeCh96, 2018-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Debian  | 
| Kernel Version (Linux Only)    |  4.11.0 | 
| Platform                                 | ARM board |
| SDK Version                            |  SDK2  | 

I want to get depth frames and process by ARM board with Debian system. I have followed [this guide](https://github.com/IntelRealSense/librealsense/issues/980) and  build `pyrealsense2.so` in my ARM board. Then I run the following code in python:

> import pyrealsense2 as rs
> pipeline = rs.pipeline()
> pipeline.start()

After several seconds, I get 391 frames, but an `Segmentation fault` error occurs:

> 372
> 373
> 374
> 375
> 376
> 377
> 378
> 379
> 380
> 381
> 382
> 383
> 384
> 385
> 386
> 387
> 388
> 389
> 390
> 391
> Segmentation fault

I run my code again and get `Segmentation fault` immediately (I get no frame this time). 
Then I plug out SR300 and plug it in, then run my code. After about 300 frames, the `Segmentation fault` error occurs.

KeCh96, 2018-06-04:
According to [this guide](https://github.com/IntelRealSense/librealsense/issues/10), this error may caused by multiple instances of the same device on Linux. But I only use one SR300. @sgorsten 

According to [this issue](https://github.com/IntelRealSense/librealsense/issues/885), we can use the following code to avoid this bug:

```
            if (
                    !ir_frame || ir_frame.get_data() == NULL ||
                    !color_frame || color_frame.get_data() == NULL ||
                    !depth_frame || depth_frame.get_data() == NULL) {

                --i;
                continue;
            }
```
So I write my code as following:

```
pipeline = rs.pipeline()
pipeline.start()

i=0
while True:
    try:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        depth_data = depth.as_frame().get_data()
        i=i+1
        print i
    except:
        continue
```

But the `Segmentation fault` error still occurs. 

Could someone provide some guidance? Thanks in advance.

KeCh96, 2018-06-04:
I found the following annotation in `/third_party/sqlite/sqlite3.h`. This information may reveal the cause of this bug. But I still don't know how to solve my problem.

```
**
** ^Calling sqlite3_free() with a pointer previously returned
** by sqlite3_malloc() or sqlite3_realloc() releases that memory so
** that it might be reused.  ^The sqlite3_free() routine is
** a no-op if is called with a NULL pointer.  Passing a NULL pointer
** to sqlite3_free() is harmless.  After being freed, memory
** should neither be read nor written.  Even reading previously freed
** memory might result in a segmentation fault or other severe error.
** Memory corruption, a segmentation fault, or other severe error
** might result if sqlite3_free() is called with a non-NULL pointer that
** was not obtained from sqlite3_malloc() or sqlite3_realloc().
**
```

KeCh96, 2018-06-04:
To analyze this error, I show my 'cmake' and 'make' output as follow:

When I run `cmake ../ -DFORCE_LIBUVC=true -DBUILD_PYTHON_BINDINGS=true`, I get:
```
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info: REALSENSE_VERSION_STRING=2.11.1
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
CMake Warning at CMakeLists.txt:576 (message):
  Using libuvc!

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for one of the modules 'libusb-1.0'
CMake Warning (dev) at /usr/share/cmake-3.11/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/aarch64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/aarch64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/aarch64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  examples/CMakeLists.txt:21 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
CMake Warning (dev) at /usr/share/cmake-3.11/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/aarch64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/aarch64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/aarch64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  tools/CMakeLists.txt:26 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.22.29
-- Found PythonInterp: /usr/bin/python3.6 (found version "3.6.5") 
-- Found PythonLibs: python3.6m
-- Performing Test HAS_CPP14_FLAG
-- Performing Test HAS_CPP14_FLAG - Success
-- pybind11 v2.2.1
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- LTO enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/linaro/kcheng/librealsense/build
```

KeCh96, 2018-06-04:
When I run `make -j4 -i`, I get:
```
Scanning dependencies of target pybackend2
[  1%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/backend.cpp.o
[  1%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/win/win-helpers.cpp.o
[  1%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/win/win-uvc.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  2%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/pybackend_extras.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  2%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/pybackend.cpp.o
Scanning dependencies of target realsense-file
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  2%] Building C object third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o
In file included from /usr/include/aarch64-linux-gnu/bits/libc-header-start.h:33:0,
                 from /usr/include/stdint.h:26,
                 from /usr/lib/gcc/aarch64-linux-gnu/7/include/stdint.h:9,
                 from /home/linaro/kcheng/librealsense/third-party/realsense-file/lz4/lz4.h:339,
                 from /home/linaro/kcheng/librealsense/third-party/realsense-file/lz4/lz4.c:91:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
[  3%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/console_bridge/src/console.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  4%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/win/win-usb.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  5%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/debug.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  5%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/win/win-backend.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  6%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/linux/backend-v4l2.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  6%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/linux/backend-hid.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  7%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/types.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  7%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/header.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  8%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  9%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/win/win-hid.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  9%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/archive.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[  9%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag_player.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
In file included from /home/linaro/kcheng/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/bag_player.cpp:1:0:
/home/linaro/kcheng/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/bag_player.h:81:42: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   BagPlayer(const std::string &filename) throw(BagException);
                                          ^~~~~
/home/linaro/kcheng/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/bag_player.cpp:8:48: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 BagPlayer::BagPlayer(const std::string &fname) throw(BagException) {
                                                ^~~~~
In file included from /home/linaro/kcheng/librealsense/third-party/realsense-file/boost/boost/function/detail/maybe_include.hpp:18:0,
                 from /home/linaro/kcheng/librealsense/third-party/realsense-file/boost/boost/function/detail/function_iterate.hpp:14,
                 from /home/linaro/kcheng/librealsense/third-party/realsense-file/boost/boost/preprocessor/iteration/detail/iter/forward1.hpp:52,
                 from /home/linaro/kcheng/librealsense/third-party/realsense-file/boost/boost/function.hpp:64,
                 from /home/linaro/kcheng/librealsense/third-party/realsense-file/rosbag/roscpp_traits/include/ros/message_event.h:42,
[... cut ...]
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
[ 99%] Linking CXX executable rs-depth-quality
/usr/bin/ld: /usr/local/lib/libglfw3.a(vulkan.c.o): undefined reference to symbol 'dlsym@@GLIBC_2.17'
//lib/aarch64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[ 99%] Built target rs-depth-quality
[100%] Linking CXX shared library pyrealsense2.cpython-36m-aarch64-linux-gnu.so
[100%] Built target pyrealsense2
```

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
hi @KeCh96,

we can reproduce this issue on our board, can you please tell which board  you are using

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
correction: we CANNOT reproduce.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @KeCh96,
can you still reproduce this issue?

closed 2018-08-21 by RealSense-Customer-Engineering