realsenseai/librealsense #1824: Realsense can not work in usb3 port

linzihao1999, 2018-06-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | unknow | 
| Operating System & Version |   Linux   | 
| Kernel Version (Linux Only)    | 4.4.55                                       | 
| Platform                                 | arm board  |
| SDK Version                            |  2.11.1                       | 

### Issue Description
Realsense can not work in usb3 port, but it work in usb2 port.

dorodnic, 2018-06-04:
Hi @linzihao1999 
What is the exact platform you are using?

jlgarcia75, 2018-06-18:
Hello @linzihao1999,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

closed 2018-06-21 by RealSense-Customer-Engineering