realsenseai/librealsense #1826: D460 Device?

dheera, 2018-06-04:
[image]

I can't find any information about a 460 device online. Can you please post specs and availability?

dorodnic, 2018-06-04:
Currently only D415 and D435 are certified to be sold as consumer products. 
We offer a range of other configurations to select customers for integration in their products.
Do you have a specific need for your project?

dheera, 2018-06-04:
@dorodnic Definitely would be interested in other configurations. Will follow up with you by e-mail.

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
hi @dheera,

do you have intel Sales channel to contact with?
or there's way for us to reach you?
thanks.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @dheera,

that's great, do you have any follow up question or let's close this ticket?

fiorano10, 2018-06-18:
I have the same question, and we do have a specific need for a project.

agrunnet, 2018-06-18:
Please contact intel sales and they can help you with getting hands on the d460.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @fiorano10,

would like to confirm if you reach our sales persons?

timprepscius, 2018-07-06:
Hey there, is there any information about this?  I tried to contact sales and got a "no information exists" reply.

dorodnic, 2018-07-11:
We appreciate the interest. 
That said, I'm closing the GitHub issue, since GitHub is not really the platform we want to use for product updates and announcements. We have a lot of information at [realsense.intel.com](https://realsense.intel.com/) and we are trying to continuously improve the site. Please stay tuned for exciting new RealSense products :)

closed 2018-07-11 by dorodnic