realsenseai/librealsense #1827: Remove IR pattern preset does not work in D435 with latest firmware

dheera, 2018-06-04:
[image]

Running firmware 05.09.11.00, tested on Ubuntu 16.04 and 18.04

dorodnic, 2018-06-04:
Hi @dheera 
This hardware feature is currently only supported by the D415 cameras.

dheera, 2018-06-04:
I see. Any plans to support it on D435s soon?

agrunnet, 2018-06-05:
Unfortunately the IR removal is not possible for D435 as the technique requires that the stereo sensors are color and not monochrome.

AntoineWefit, 2018-06-06:
Out of curiosity, may I ask what the "Remove IR Pattern" should do ? Is it supposed to remove the IR pattern in the output stream ?
I tried it with the latest versions of the D415 firmware and Intel RealSense Viewer, but It doesn't seem to do anything at first sight (unless I missed something ?).

I didn't find much explanation on this preset, excepted maybe this page [https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets](url) (which contain a "broken ?" link to the "D415_RemoveIR.json"). 
You actually can access the file by typing [https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/D415_RemoveIR.json](url), but the preset seems to be a bit empty compared to the one provided when saving the "Remove IR Pattern" preset from the RealSense Viewer. I saw that only the "aux-param-colorcorrectionX" parameters are modified.

I also assume that the removal of the IR pattern is not exclusively controlled by a ".json" file, so how does the preset behave exactly ?

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
hi @AntoineWefit,

do you mean when you load remove IR pattern preset the D415 output is the same? due to the IR projector is not too strong, so maybe you need to target to a near object like 30cm distance. if so please provide your screenshot so we can take a look.
and remove IR.json provide parameters modified so much less parameters compared to file you saved from Realsense Viewer, it will save almost all parameters

AntoineWefit, 2018-06-13:
@RealSense-Customer-Engineering 
> do you mean when you load remove IR pattern preset the D415 output is the same? 

Yes, it seems so.
(please don't pay attention to the actual acquisition, the camera position is random atm)

With the "Remove Ir Pattern" Preset
[image]

With the "Default" Preset
[image]

Look pretty identical to me :/

Could you specify what is the expected output with the "Remove Ir Pattern" Preset ?

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
hi @AntoineWefit,

sorry, i am wrong, you should use D415 with FWv5.9.11, in Realsense viewer set available streams: Infrared1 forma to RGB8, then you will see the "Remove IR Pattern" Preset works when you load it.

AntoineWefit, 2018-06-25:
Thank you @RealSense-Customer-Engineering !
It's working great !

One last question though, is it possible to disable the pattern itself with some parameters (software side) or is it an intrinsic part (hardware side) of the IR projector (I guess some kind of "lens") ? It's not that much of a big deal, but I'm curious.

agrunnet, 2018-06-25:
We would love to have a cheap projector that can switch between a pattern and a uniform illumination, but that does not currently exist. So the one we have is a fixed pattern. You can only turn it on and off and change the power.

AntoineWefit, 2018-06-25:
Ok good to know, thank you for your answer.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @AntoineWefit,

seems all your questions are answered, can we close this ticket now?

AntoineWefit, 2018-07-02:
@RealSense-Customer-Engineering 
Yes indeed !
But this is not my thread, I can't close it.

closed 2018-07-02 by dorodnic