realsenseai/librealsense #1828: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

anshulvj, 2018-06-04:
Camera mode: RS415
OS: Ubuntu 16.04 64-bit
Language: Using librealsense python wrapper for data acquisition and processing.

I am trying to debug my Python code that is retrieving point cloud information from realsense D415 sensor. The sensor usually works fine but during debug if I have to quit debugging and restart the debug process it says:

`xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy`
`Process finished with exit code 0`

For proprietary issues I cannot put my code here but essentially it's acquiring x,y,z information from the sensor and computing various things on that data.

This usually happens during debugging the code. Has anyone faced similar issue before? 

Any help is appreciated. Please let me know if you need further information.

xufeifeiWHU, 2018-06-05:
This problem occurs when the device is really busy now, I means that there are several occasions it will be busy:
1.if you run/debug your codes in a python IDE (ie.pycharm..) and have creat a pipline then use the frame you get, but have you realse your camera in your codes ? or did your project exit/stop normally? 
2.Please check carefully if there are another project was using your camera
3.If you run your  .py file in command line, the python3.5/2.7 will help you release camera but please keep in mind Please release your camera mannually, you can find such demo codes in the wrapper/examples.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
@anshulvj
does the suggestion provided by @xufeifeiWHU work for you?

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
assuming the issue is solved, close the issue first, if you have related problem, please reopen it or create new issue.

closed 2018-06-19 by RealSense-Customer-Engineering

lchop, 2018-08-27:
Hello,

I have the same error. How do you realse the camera ?
Best regards

eyildiz-ugoe, 2019-08-09:
I had the same problem as I was running the ROS node for realsense, as I terminated it, the problem was gone.

jebasamuel, 2019-12-02:
@anshulvj @lchop @eyildiz-ugoe 
The probem is kernel prevents this dynamic resolution setting for same device from different application.However, if both application have same resolution setting, this can be relaxed. buf, if the streaming is already happening, you will get this error. 

This can be solved by applying this kernel patch.

https://patchwork.kernel.org/patch/11179557/

locdoan12121997, 2020-02-28:
@jebasamuel 
Hi, I have apply the kernel patch but still not work. What I did to apply the patch was:
Download patch https://patchwork.kernel.org/patch/11179557/
Then find the file name hantro_v4l2.c, which appear to be in ~//librealsense/ubuntu-bionic-5/drivers/staging/media/hantro/hantro_v4l2.c, and direct to the directory
Then, patch < media-hantro-relax-s_fmt-out-busy-error.patch 
But I still have the same xioctl error

jebasamuel, 2020-02-28:
@locdoan12121997 
You need to apply the patch for correct v4l2 driver code. Check your v4l2 driver. it may not be necessary same as the hantro.

dmoham1476, 2021-05-04:
I am seeing this issue on jetson with realsense especially when the client pings a realsense server repeatedly. I followed instructions to install realsense on jetpack using this blog - https://lieuzhenghong.com/how_to_install_librealsense_on_the_jetson_nx/
How can I apply the patch?