realsenseai/librealsense #1830: Where is the source code for the alignment function in RealSense SDK2.0?

ToshiEAB, 2018-06-05:
Hello,

I am trying to figure out where the source code (C++) is for the alignment function in RealSense SDK2.0 and wonder if someone can give me an advice. 

Specifically, I wrote codes for aligning a depth frame to the corresponding color frame as follows: 

rs2::frameset frames = pipe.wait_for_frames();
rs2::align align(rs2_stream::RS2_STREAM_COLOR);
rs2::frameset aligned_frames = align.process(frames);
rs2::frame color_frame = frames.get_color_frame();
rs2::frame depth_frame = aligned_frames.get_depth_frame();

I'd like to see the source code for checking how the alignment is accomplished. I right-clicked "align" and "process" shown above to see how they are declared or defined and have reached "class align{}" in "rs_processing.hpp". From there, however, I cannot find the calculation part (I am guessing calculations for projection, rotation, and deprojection take place for each pixel using for loops like the ones shown in align.cpp).

Information about my environment is as follows:

Camera model: D400
OS: Win 10
Platform: PC
SDK version: 2.10.3

Regards,

dorodnic, 2018-06-05:
[align.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp)

ToshiEAB, 2018-06-05:
Thank you! So I had already found it, hadn't I?

dorodnic, 2018-06-05:
Yes. It is all a bit confusing, we are using [hourglass interfaces](https://www.youtube.com/watch?v=PVYdHDm0q6Y) pattern for ABI compatibility (instead of something like COM), forcing us to write every C++ class twice - one time for real and second time over C API. 
This creates some challenges understanding the code, but it does a good job preventing compiler ABI related crashes and memory corruptions.

ToshiEAB, 2018-06-05:
dorodnic,

Thanks. That's good to know. I need to learn more about C++.

Regards,

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
hi @ToshiEAB,

do you need more support on this topic?

ToshiEAB, 2018-06-13:
Thanks for checking. I’m closing this thread.

closed 2018-06-13 by ToshiEAB