realsenseai/librealsense #1831: Export depth image in grey scale

baptiste-mnh, 2018-06-05:
Hi!
As the title say, I want to export the depth image in a grey scale (dynamic and white to black)
[image]

By defaut this is the result of the depth export : 
[image]

How to get the depth image as in the Realsense Viewer?
Thanks.

dorodnic, 2018-06-05:
Hi @baptiste-mnh 
I'm not sure we have this exact functionality in the Viewer (I think the Depth Quality Tool can export both RAW and "colorized" depth data to PNG)
To get this in code, you probably want to look at [save-to-disk](https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk). To customize the color-map, you can add `set_option` on the `colorizer` object:
```cpp
// Declare depth colorizer for pretty visualization of depth data
rs2::colorizer color_map;
color_map.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 1.f);
color_map.set_option(RS2_OPTION_COLOR_SCHEME, 2.f); // White to Black
```

baptiste-mnh, 2018-06-05:
Thanks it works!
For those who need to do the same just use this code to save it (more simple than in the exemple) : 
```
void DepthExporter::export_scan(std::string filename, rs2::frameset &frames) {
    rs2::colorizer color_map;
    color_map.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 1.f);
    color_map.set_option(RS2_OPTION_COLOR_SCHEME, 2.f); // White to Black

    auto depth = frames.get_depth_frame();
    cv::Mat dMat(cv::Size(1280, 720), CV_8UC3, (void*)color_map(depth).get_data());

    cv::imwrite( filename, dMat );
    confirmeSave(filename);
}

```

closed 2018-06-05 by baptiste-mnh

WeiChihChern, 2018-08-26:
Can we have a CV_8UC1 grayscale instead of CV_8UC3?

Usernamezhx, 2018-11-20:
what is the python API about the `color_map.set_option`

lramati, 2018-11-21:
It's the same API as wherever else options appear, such as on sensors.
Python's built in help() function should be able to help you

Usernamezhx, 2018-11-29:
@lramati  but I cannot find the corresponding definition about the : `RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED`