realsenseai/librealsense #1834: Unable to get depth stream from SR300

elfslmn, 2018-06-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300  | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Ubuntu 16.04 LTS | 
| Kernel Version (Linux Only)    |  4.13.0-43-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.11.1                     | 

### Issue Description
When I start the realsense viewer and connect my SR300 camera I got warnings above
 05/06 16:35:37,568 WARNING [139703620278848] (backend-v4l2.cpp:1156) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 05/06 16:35:37,569 WARNING [139703620278848] (backend-v4l2.cpp:1156) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 05/06 16:35:37,569 WARNING [139703620278848] (sensor.cpp:330) Unregistered Media formats : [ INZC PAIR Y10  YUYV ]; Supported: [ ]

Then when I try to get depth data

dorodnic, 2018-06-05:
This `failed Last Error: No such device` seems suspicious, can you try running the device via powered USB3 hub? I think the device gets disconnected when you start streaming.

elfslmn, 2018-06-05:
I tried a powered hub, it say the output is 5V-1A. it didnt worked but I am not sure if it is usb3 or not. Do you know how can I check if it is usb3 or not? The interesting thing, the camera was working yesterday without any external hub.

dorodnic, 2018-06-05:
Well, SR300 is only recognized via USB3, so it is USB3.
Did you try changing the cable?

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
Hi @elfslmn
you can use lsusb  -t or lsusb -v to get detail info of USB devices
and SR300 require USB3 connection.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
Hi @elfslmn,
is this still an issue on your setup?
you can use lsusb  -t or lsusb -v to get detail info of USB devices to know if SR300 is enumerated and  SR300 required USB3 connection
best

closed 2018-08-30 by RealSense-Customer-Engineering

RealSense-Customer-Engineering, 2018-08-30:
[Realsense Customer Engineering Team Comment]
issue is closed