realsenseai/librealsense #1842: aligning depth frame to color frame in node-librealsense

nilan, 2018-06-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                   | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04.4 LTS                            | 
| Language                               |  nodejs                                                 | 

### Issue Description

I am using nodejs wrapper for librealsense (node-librealsense). When I try to align the frames with align stream as rs2.stream.STREAM_COLOR,  depth frame content is all zero.

Below is the code snippet i am using.

const align = new rs2.Align(rs2.stream.STREAM_COLOR);
const pipeline = new rs2.Pipeline();
const profile = pipeline.start();
const colorizer = new rs2.Colorizer();

const frameset = pipeline.waitForFrames();
const alignedFrameset = align.process(frameset);
 frame = alignedFrameset.colorFrame;
 depth = alignedFrameset.depthFrame;

If I pass align stream as rs2.stream.STREAM_DEPTH, then the content of color frame becomes all zero.

Please suggest me how to align the depth and color frames.

tingshao, 2018-06-14:
@nilan I had a try on the wrappers/nodejs/examples/nodejs-align.js by just changing the 
`const align = new rs2.Align(rs2.stream.STREAM_DEPTH)` into `const align = new rs2.Align(rs2.stream.STREAM_COLOR)`, while it works. Did you also changed other code?

dorodnic, 2018-07-12:
Hi @nilan 
Any update on this? Do you still need help?