realsenseai/librealsense #1843: Issue installing librealsense module

AkhilJPatil, 2018-06-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Operating System & Version |   Win (10)  |

### Issue Description
Hello team,

I am completely new to the overall idea of working with camera and realsense in specific. And I am facing issue at the very 1st step of installing the librealsense.

I tried installing the librealsense following the steps mentioned in the node-librealsense page.
I have set all the necessary prerequisites in terms of cmake, visual studio, python2.7. All the paths are set correctly. And to the best of knowledge I have installed the correct 64bit version for the needed softwares.

However, I am getting this below error when I perform npm install.

I have tried uninstalling everything and reinstalling it again multiple times to get this thing work but still the same error reoccurs.

ERROR:

> CMake
> 
>  Error at CMakeLists.txt:4 (project):
>   Generator
> 
>     NMake Makefiles
> 
>   does not support platform specification, but platform
> 
>     x64
> 
>   was specified.
> 
> CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
> CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
> 
> 'Failed to configure librealsense'
> 
> E:\2_Unhali_sem\SP_Kray\NewLasttry\SP-RCARAP\node_modules\node-librealsense\scripts\npm_dist\build-librealsense.js:67
>       throw new Error('Failed to build librealsense, build-dist.bat exited with code ${code}');
>       ^
> 
> Error: Failed to build librealsense, build-dist.bat exited with code ${code}
>     at ChildProcess.child.on (E:\2_Unhali_sem\SP_Kray\NewLasttry\SP-RCARAP\node_modules\node-librealsense\scripts\npm_dist\build-librealsense.js:67:13)
>     at emitTwo (events.js:126:13)
>     at ChildProcess.emit (events.js:214:7)
>     at maybeClose (internal/child_process.js:925:16)
>     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! node-librealsense@0.2103.0 preinstall: `node scripts/npm_dist/build-librealsense.js`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the node-librealsense@0.2103.0 preinstall script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     C:\Users\Lenovo\AppData\Roaming\npm-cache\_logs\2018-06-06T11_03_21_144Z-debug.log
 
In the error logs below message was displayed.

> 142 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
> 143 verbose node v8.11.1
> 144 verbose npm  v5.6.0
> 145 error code ELIFECYCLE
> 146 error errno 1
> 147 error node-librealsense@0.2103.0 preinstall: `node scripts/npm_dist/build-librealsense.js`
> 147 error Exit status 1
> 148 error Failed at the node-librealsense@0.2103.0 preinstall script.
> 148 error This is probably not a problem with npm. There is likely additional logging output above.
> 149 verbose exit [ 1, true ]

Tapi2508, 2018-06-06:
Ach

AkhilJPatil, 2018-06-06:
It may not be an issue as such but could find anyone facing this issue on other forums and groups. Any help would be appreciated by my team even if it may turn out to be a trivial issue. :)

tingshao, 2018-06-14:
Hi, @AkhilJPatil That seems to be an env issue. Did you use the **Developer Command Prompt for VS xxx**? That prompt should have setup the dev env for you. I used **Developer Command Prompt for VS 2017**.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
@AkhilJPatil
is the issue still there?

closed 2018-07-06 by RealSense-Customer-Engineering