realsenseai/librealsense #1845: Visual C++ Redistributable Required?

jb455, 2018-06-06:
Hi,

We have developed a C#/WPF app which uses the librealsense wrapper. The app references the wrapper, and the installer copies the native realsense2.dll library to the app install directory so the wrapper can find it on a machine which doesn't have the sdk installed. This works fine most of the time, but on some machines we get an "Unable to load DLL" exception when any camera functions are attempted by the app. This problem is immediately fixed when the Visual C++ 2017 Redistributable is installed. Is the redistributable a prerequisite for using librealsense? Interestingly, the Intel.Realsense,Viewer app works ok without the redistributable so I guess maybe it's something to do with the wrapper as the Viewer is C++?

Thanks,
James

dorodnic, 2018-06-06:
Hi @jb455 
Did you try the pre-compiled binaries we provide in [Intel.RealSense.SDK.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.12.0/Intel.RealSense.SDK.exe)?

jb455, 2018-06-06:
Hi,
I'm pretty sure it is the precompiled native DLL we're distributing, though maybe not the precompiled wrapper. I should've said before, it is an older version of librealsense we're currently using as we haven't updated it in a couple of months, so if you've changed anything recently this may no longer be relevant. 
I just wanted to check if it's known that we need to do this all the time (so we'll have to bundle the redist with our app installer) and most of the PCs we've installed it on have already had the redist, or something else is going on with the few isolated cases of the app not working straight off.

Would you expect precompiled binaries to always be fine but binaries I compile myself (using Visual Studio) to need the redist? That would be an easy fix as I think everything extra I'd added to the wrapper is now in the main codebase so I won't need to compile my own any more.
Thanks

RealSense-Customer-Engineering, 2018-06-20:
[Realsense Customer Engineering Team Comment]
@jb455
Does it answer your question?

jb455, 2018-06-20:
Erm, yes to some extent.. Do the precompiled binaries have the dependency or not? Also, does turning off STATIC_CRT have any performance impact as turning on OPENMP does (or any other impact - eg, does it package all the common libraries it usually gets from the redist, increasing the dll file size)?
It would probably be worth drawing attention to the dependency in the docs somewhere as it took me a couple of hours of tearing my hair out before I chanced upon installing VC++ fixing it, so making it clear would save others some stress!

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
@jb455
1. For the precompiled binaries, the required VC dependency is inside it, right?

Correct, the binaries in the installer are compiled with static CRT (static MS libraries), and therefore is self-contained and does not require MS redistribution package.

2. Any performance impact with compiler option “STATIC_CRT” being turned on/off?

     It should not, but since it is a 3rd party library (Microsoft) and we have no control over it.

3. How's the binary size with STATIC_CRT on/off?

In a test I did librealsense2 dll is 6.8Mb with “STATIC_CRT”  turned off and 8.2Mb otherwise.

BTW, just found out I have typo in previous response, if you need the static link you need to turn **ON** the STATIC_CRT instead of turning of, sorry for the confusion.

jb455, 2018-06-21:
Great, thanks for the clarifications. Do you intend to add this information into the documentation? Shall I leave this issue open until that's done or close it now?

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
@jb455
Please help to close it first, I will track internally when will it be documented.

closed 2018-06-21 by jb455