realsenseai/librealsense #1846: Using Intel Real sense SDK in Unity - Cameras matching

ThibaultLesaunier, 2018-06-06:
| Required Info                         |            What I have                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |  Win 8.1 | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C#/unity }                          | 

### Issue Description
Hi everyone,

I'm working on an app using D415 realSense camera and the Unity wrapper of Intel Real Sense sdk 2.0.
I have two questions for you:

Is it possible to interact with the SDK from Unity using the wrapper or something else ? 

Is it possible to match the Unity Camera on the D415 with the SDK? (Augmented reality problem)

Thanks. :-)

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
Hi @ThibaultLesaunier,

Did you run any issue using the Unity wrapper in librealsense? Suggest to use the latest librealsense (2.12.0) which fixes some issues.
Can you clarify about "Augment reality problem"?

ThibaultLesaunier, 2018-06-08:
Hi,

Thanks for the answer. 

I mean, is it possible to use the RealSense SDK from the Unity wrapper ? Asking C++ code from C# in Unity or accessing to the library from somewhere ^^.

My interest on your Unity wrapper for IntelRealsense Sdk is to use RGB and Depth of the camera to produce an augmented reality app in Unity. So I'm looking for a solution to match the RealSense Camera with my Virtual Camera in Unity, I can do that by my way but maybe you already have something to do it.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
Hi @ThibaultLesaunier,

So far, only C# wrapper supported. Need to further check internally about C++ wrapper and Virtual Camera in Unity as feature requests.
You can also discuss in the realsense community if anyone already has a better solution/idea.

RealSense-Customer-Engineering, 2018-07-31:
[Realsense Customer Engineering Team Comment]
Hi @ThibaultLesaunier,

Please try librealsense 2.14.0 which includes "AlignmentSample.unity" to see if it can fulfill your request.

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
Hi @ThibaultLesaunier,

Did you try the latest Unity wrapper and see if it is ok?

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
HI @ThibaultLesaunier,

I will close this issue first. If you try the latest Unity wrapper and still have any concern, still can file an issue or reopen it.

closed 2018-08-24 by RealSense-Customer-Engineering