realsenseai/librealsense #1849: RealSense error calling rs2_load_json

omerbrandis, 2018-06-07:
dear support,

i've just upgraded librealsense on my pc from v 5.10.1 to 5.12

i'm getting the following error ( software used to work in previous version)

RealSense error calling rs2_load_json(dev:0xe957a0, json_content:0xe957e0, content_size:3513):
    get_xu(id=2) failed! Last Error: Input/output error

RealSense error type Invalid Value.

i'm using d415,with the latest firmware (5.9.11).

i haven't changed my code or the input file.

here is the code:
  rs400::advanced_mode MyDevice=MyPipelineProfile.get_device();
    MyDevice.load_json(...file name....);

i've tried creating a new json config file using realsense-viewer , default settings -> download , 
but this didn't fix the problem.

the viewer is also unable to use a config file it generates. 
( it can load the file, but then opening the rgb stream failes, with an error message about an invalid profile).

please advise.
Omer.

omerbrandis, 2018-06-07:
i've changed the logging level to info , 
here is the data printed

 07/06 13:32:18,870 INFO [140471706744640] (environment.cpp:85) Found 4 unreachable streams, 1740 extrinsics deleted
 07/06 13:32:18,958 INFO [140471706744640] (backend-v4l2.cpp:624) Trying to configure fourcc Z16 
 07/06 13:32:18,963 INFO [140471706744640] (backend-v4l2.cpp:624) Trying to configure fourcc YUYV
 07/06 13:32:18,968 WARNING [140471519139584] (backend-v4l2.cpp:866) Empty frame has arrived.
 07/06 13:32:18,968 WARNING [140471519139584] (backend-v4l2.cpp:866) Empty frame has arrived.
 07/06 13:32:18,968 WARNING [140471519139584] (backend-v4l2.cpp:866) Empty frame has arrived.
 07/06 13:32:18,968 WARNING [140471519139584] (backend-v4l2.cpp:866) Empty frame has arrived.
 07/06 13:32:19,408 INFO [140471519139584] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,408 INFO [140471519139584] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,408 INFO [140471519139584] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,408 INFO [140471519139584] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,503 INFO [140471527532288] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,503 INFO [140471527532288] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,503 INFO [140471527532288] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,503 INFO [140471527532288] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:19,575 INFO [140471519139584] (metadata-parser.h:236) frame_number - last_frame_number 0
 07/06 13:32:26,594 WARNING [140471706744640] (types.cpp:57) get_xu(id=2) failed! Last Error: Input/output error

hope this helps

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

would like to confirm that you are totally fine with 2.10.1 and not working with v2.12.0?
do you mean Realsense viewer save a new JSON file and can't load it back? or Realsense viewer can't read old JSON file?
i would like to reproduce this, so need to know which OS you are using?

omerbrandis, 2018-06-08:
Dear realsense customer engineering team.

1. I had a few problems with v2.10.1, and was instructed to upgrade.

2. I am now working with 2.12,
It did not solve the original problem i faced before
And it introduced the new problem that i described in this thread.

3. I am usung ubuntu 16.04

4. Realsense viewer cant load back a file that it itself has gendrated in version 2.12.
( file loads but later cant open the rgb stream).

+V5.12 Viewer "cant load" a file generated with 5.10.1 viewer.

+ my cpp code fails in r2_load_json for files generated with both the v5.10.1 and v
5.12 viewer.

Omer.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,
I think we should start from your #4. my realsense viewer from v2.12.0, can save JSON file and read it back.
and then RGB can't open is an known bug listed on v2.12.0, (DSO-9543).
you can enable Realsense viewer-->RGB camera-->Availale Streams-->Color, then you should able to start RGB streaming with your setting.
please let me know how it works, thanks.

omerbrandis, 2018-06-10:
Dear realsense customer engineering team.

1. I was not able to find details about the known bug DSO-9543.
can you please provide a link to detail information?

2. I did not understand your suggestion -
"you can enable Realsense viewer-->RGB camera-->Availale Streams-->Color, then you should able to start RGB streaming with your setting."

**my real problem is not being able to use function  rs2_load_jsonj ,** 
I added the description for the error in the "VIEWER" because i guessed that they are caused by the same problem, and it's an easy way to prove that the error is in librealsense and not in my code.
( i don't require a workaround for the viewer's inability to load a json file )

thanks,
Omer.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

please refer to https://github.com/IntelRealSense/librealsense/pull/1888.
please let me know how it works, thanks.

omerbrandis, 2018-06-27:
dear  Customer Engineering Team

1) i've reviewed #1188, but i'm afraid i did not understand it .
on the one hand  : " Root cause = Loading/Saving settings currently only supports stereo stream (Depth,IR)"
on the other " The fix: Disable only Depth/IR streams upon load settings and enable only requested streams afterwards"

this seems to contradict itself.

2) is the suggested solution a workaround ?
or will the api change permanently ?
( i would rather avoid performing a software change to workaround the problem.
i already have a simpler workaround - downgrading libreal sense to version that works )

thanks,
Omer.

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

the idea is to verify if Realsense viewer can read jason file at your side to verify if there's environment issue, where Realsense viewer from libRealsense v2.12.0 have known issue when reading jason file which fixed by pull request #1888.
so if Realsense viewer is ok to load jason file it generated, we can make sure the environment and .json file is no problem, so would suggest you to have a try with libRealsense v2.13.0 (have pull request #1888 integrated), to see if realsenxe viewer can load json file and if your code works too.

for your question 2), if you are referring pull request #1888, it's permanent fix.
and good to know you have a workaround so this issue not blocking you.

please let me know how it works, thanks.

omerbrandis, 2018-07-10:
hi

I intended to close this issue after i have tested v 2.13.0 and made sure it solves the problem.

but i will not be able to do this in the near future .

Thanks,
Omer.

closed 2018-07-20 by omerbrandis

mprat, 2018-09-20:
I am having this exact same issue using librealsense v2.16.0 and Firmware version 5.10 on a D415 and a TX2 tegra device.

What is the fix?