realsenseai/librealsense #1850: No realsense devices found for samples

lasti15, 2018-06-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic      | 
| Platform                                 | PC  |
| SDK Version                            |   2    | 

### Issue Description
I am trying to set up my camera for the first time and play around with the samples.
I have build the SDK from source using: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
And can successfully view both the 2D and 3D streams on realsense-viewer and rs-capture.

However, when I try to run a sample, e.g. the or_pt web tutorial presented here: https://software.intel.com/sites/products/realsense/samples/md_samples_or_pt_tutorial_1_web_README_or_pt_tutorial_1_web.html

I get the message:
Error: Device is null. There are no RealSense devices connected.
Please connect a RealSense device and restart the application
Segmentation fault (core dumped)

What is the problem here? Why would rs-capture be able to detect the camera but not rs_or_pt_tutorial_1?

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @lasti15,

the or_pt web tutorial  is sample for libRealsense 1 and you are using libRealsense 2 now, so that sample is not working.

lasti15, 2018-06-08:
Are there not any equivalent or/pt modules in libRealsense 2? That seems like quite a downgrade?

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @lasti15,.
Our SDK is focused on providing the best depth data across multiple wrappers and operating systems. We wish to grow our open source community and encourage them to do something wonderful with our products. We have no plan to add middleware libraries to our product but we may partner with 3rd parties in the future.

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
hi @lasti15,

do you need more support on this topic?

closed 2018-07-12 by dorodnic