realsenseai/librealsense #1851: Error in RS Viewer, loading JSON: "Extension control must be initialized before use"

xtrawurst, 2018-06-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version | Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.11.1 | 

I am trying to load the attached JSON (no_SLO.json) in the RS Viewer and get this error message:

"Extension control must be initialized before use"

Also see attached screen shot.

My intention was to load a different JSON file, then load the attached JSON "on top". I expected all the values of the old file to persist and only the two values of the attached file (no_SLO.json) to be changed. My expectation was due to this thread:
https://github.com/IntelRealSense/librealsense/issues/1815

It seems that my mental model of how the preset system works is still wrong.

Therefore I would really, **really** appreciate an explanation of how I can understand the preset system to work after the recent FW change that made things persistent.
Some things seem to persist, others don't. Some things seem to be dependent on other things, but I'm in the dark about this.

[image]

[no_SLO.json.txt](https://github.com/IntelRealSense/librealsense/files/2080354/no_SLO.json.txt)

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

would like to confirm your issue, due to I tried to load no_SLO.json from you and I can load it without error with Realsense Viewer v2.12.0
though I tested with D435, but I think it should be the same, will try with D415 later.
checked my realsense viewer, applied your JSON file do modify 2 setting only, and from my test, it set other values back to default.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
correction, with more test, I found other values from set to default, continue testing on this.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
confirmed my test result, values only changes if JSON file set the value, other values no change. will not set those value to default.

xtrawurst, 2018-06-08:
@RealSense-Customer-Engineering 
Thanks. I will update my RS Viewer to 2.12.0 and try this again.

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

would like to know if your test with RS viewer v2.12.0 is good?

closed 2018-07-23 by RealSense-Customer-Engineering