realsenseai/librealsense #1852: Unity Wrapper Stream Aligner does not align Infrared Stream

CodaxD, 2018-06-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 
| Firmware Version                   | 05.08.15.00 
| Operating System & Version |  Windows10  
| Kernel Version (Linux Only)    | n/a
| Platform                                 | PC Unity 2018.1.2
| SDK Version                           | 2.0 Unity
| Language                               |  C#, Unity Wrapper 

### Issue Description
I am trying to use the Realsense Align Component to align the infrared stream to the depth stream like in the Realsense viewer.  However attempting to change the stream from "Color" to "Infrared" returns a null 'colorFrame' 

From PointCloudGenerator.cs
    
    void Start()
    {
        aligner = new Align(Intel.RealSense.Stream.Infrared);
        RealSenseDevice.Instance.onNewSampleSet += OnFrames;
    }

    private void OnFrames(FrameSet frames)
    {
        using (var aligned = aligner.Process(frames))
        {
            using (var colorFrame = aligned.ColorFrame)
            using (var depthFrame = aligned.DepthFrame)
            {
                ....

colorFrame is null when Stream is Set to "Infrared"

This is also repeatable in AlignImages.cs

Expected Results:
Align Class should work on Infrared Stream and Color Streams

Please let me know If there is another way to align the Infrared and Depth Streams.  Thanks!

CodaxD, 2018-06-07:
For now, I'm getting good enough results from shifting the depth image along the X axis
[image]

dorodnic, 2018-06-11:
Hi @CodaxD 
When you are aligning to Infrared the result would not have a `ColorFrame`. Currently we don't provide an `InfraredFrame` helper function on the `FrameSet`, but if you look at what [ColorFrame function is doing](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/FrameSet.cs#L59) it should be easy enough to implement something similar on top of the API:
```
VideoFrame infrared = null;
foreach (var frame in aligned)
{
    if (frame.Profile.Stream == Stream.Color)
         infrared = frame as VideoFrame;
    frame.Dispose();
}
```
Please let us know if this can help, and we will add `InfraredFrame` extension in one of the future releases.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Hi @CodaxD,

Is above answer resolve your development issue?

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
Hi @CodaxD,

Any update about @dorodnic's suggestion? or, still wait for our official release?

dorodnic, 2018-11-08:
I believe the new Unity package has everything you need. If you have any follow-up questions, feel free to open another issue or ping this one

closed 2018-11-08 by dorodnic