realsenseai/librealsense #1854: realsense2_camera rs_multiple_devices.launch fails to launch two D415 cameras

ankurbaruah90, 2018-06-08:
Hi,

I am using two D415 cameras and I wish to subscribe to data from both of them. To launch both cameras together I used the rs_multiple_devices.launch file, however the launch file fails to launch the cameras. On the other hand if I launch the devices individually, they seem to run just fine. Suspecting it to be an issue related to devices trying to access the RealSense library files simultaneously, I tried to use a script to launch the devices separately with an acceptable delay (15 sec) in between, but the same error turns up. I am getting the following error messages:
```
(types.h:179) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
(types.h:179) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
(backend-v4l2.cpp:1243) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
(backend-v4l2.cpp:1243) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
(types.h:179) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
(backend-v4l2.cpp:1243) xioctl(VIDIOC_DQBUF) failed Last Error: No such device
```
On other instances I have also encountered `get_xu(id=11) failed! Last Error: Input/output error` when attempting to start multiple cameras. I am posting the RealSense log in debug level here.
```
SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 1000
 * /camera1/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye: True
 * /camera1/realsense2_camera/enable_imu: True
 * /camera1/realsense2_camera/enable_infra1: True
 * /camera1/realsense2_camera/enable_infra2: True
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_height: 480
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera1/realsense2_camera/fisheye_width: 640
 * /camera1/realsense2_camera/gyro_fps: 1000
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera1/realsense2_camera/infra1_fps: 30
 * /camera1/realsense2_camera/infra1_height: 480
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera1/realsense2_camera/infra1_width: 640
 * /camera1/realsense2_camera/infra2_fps: 30
 * /camera1/realsense2_camera/infra2_height: 480
 * /camera1/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera1/realsense2_camera/infra2_width: 640
 * /camera1/realsense2_camera/json_file_path: 
 * /camera1/realsense2_camera/serial_no: 805212060147
 * /camera2/realsense2_camera/accel_fps: 1000
 * /camera2/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera2/realsense2_camera/align_depth: False
 * /camera2/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/color_fps: 30
 * /camera2/realsense2_camera/color_height: 480
 * /camera2/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera2/realsense2_camera/color_width: 640
 * /camera2/realsense2_camera/depth_fps: 30
 * /camera2/realsense2_camera/depth_height: 480
 * /camera2/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
[... 173 lines cut ...]
 * /rosversion: 1.12.7

NODES
  /camera2/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)
  /camera1/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311
rs_multiple_devices.launch http://localhost:11311
core service [/rosout] found
process[camera1/realsense2_camera_manager-1]: started with pid [1310]
process[camera1/realsense2_camera-2]: started with pid [1311]
process[camera2/realsense2_camera_manager-3]: started with pid [1312]
process[camera2/realsense2_camera-4]: started with pid [1314]
[camera2/realsense2_camera-4] killing on exit
[camera2/realsense2_camera_manager-3] killing on exit
[camera1/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
It would be of great assistance if you could look into the issue. Thanks.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @ankurbaruah90,

I see you are using older libRealsense and FW versions, would you mind to try with latest version?

ankurbaruah90, 2018-06-11:
Hi @RealSense-Customer-Engineering, thanks for your revert. I have updated the librealsense and FW versions to v2.12.0 and v5.9.11 respectively, however the problem still persists. This is the current RealSense log.

```
SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 1000
 * /camera1/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye: True
 * /camera1/realsense2_camera/enable_imu: True
 * /camera1/realsense2_camera/enable_infra1: True
 * /camera1/realsense2_camera/enable_infra2: True
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_height: 480
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera1/realsense2_camera/fisheye_width: 640
 * /camera1/realsense2_camera/gyro_fps: 1000
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera1/realsense2_camera/infra1_fps: 30
 * /camera1/realsense2_camera/infra1_height: 480
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera1/realsense2_camera/infra1_width: 640
 * /camera1/realsense2_camera/infra2_fps: 30
 * /camera1/realsense2_camera/infra2_height: 480
 * /camera1/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera1/realsense2_camera/infra2_width: 640
 * /camera1/realsense2_camera/json_file_path: 
 * /camera1/realsense2_camera/serial_no: 
 * /camera2/realsense2_camera/accel_fps: 1000
 * /camera2/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera2/realsense2_camera/align_depth: False
 * /camera2/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera2/realsense2_camera/color_fps: 30
 * /camera2/realsense2_camera/color_height: 480
 * /camera2/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera2/realsense2_camera/color_width: 640
 * /camera2/realsense2_camera/depth_fps: 30
 * /camera2/realsense2_camera/depth_height: 480
 * /camera2/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera2/realsense2_camera/depth_width: 640
 * /camera2/realsense2_camera/enable_color: True
 * /camera2/realsense2_camera/enable_depth: True
 * /camera2/realsense2_camera/enable_fisheye: True
 * /camera2/realsense2_camera/enable_imu: True
 * /camera2/realsense2_camera/enable_infra1: True
 * /camera2/realsense2_camera/enable_infra2: True
 * /camera2/realsense2_camera/enable_pointcloud: False
 * /camera2/realsense2_camera/enable_sync: False
 * /camera2/realsense2_camera/fisheye_fps: 30
[... 88 lines cut ...]
[ INFO] [1528709053.267198405]: Expected frequency for color = 30.00000
[ INFO] [1528709053.298517528]: setupStreams...
[ INFO] [1528709053.303278679]: setupStreams...
[ INFO] [1528709053.308805943]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.309167196]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.309468945]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.315237304]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.315588281]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.315883607]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.321435300]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1528709053.325442260]: color stream is enabled - width: 640, height: 480, fps: 30
 11/06 14:54:13,381 ERROR [140458078795648] (types.h:179) xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[ERROR] [1528709053.387862041]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[ERROR] [1528709053.390398288]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[ INFO] [1528709053.396935284]: publishStaticTransforms...
[ INFO] [1528709053.402539731]: RealSense Node Is Up!
[FATAL] [1528709053.508434502]: Failed to load nodelet '/camera2/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[... cut ...] a2-realsense2_camera-5.log].
log file: /home/nuc/.ros/log/30afb140-6d59-11e8-8a61-94c69112c2eb/camera2-realsense2_camera-5*.log
 11/06 14:54:18,423 WARNING [140629448890240] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1528709058.426168584]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error: 
```
Is it a problem with the ROS package where it is unable to handle multiple accesses to the RealSense library simultaneously?

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @ankurbaruah90,

Did you try to use "realsense-viewer" or rs-multicam" app to see if two D415 cameras run normally on your test platform?
Are two cameras on separate USB3 ports or connected to a USB hub? Can you also try below command and show me the logs?
$ lsusb
$ usb-devices

I will also check if I can reproduce the issue.

ankurbaruah90, 2018-06-21:
Hi @RealSense-Customer-Engineering,

I have used both "realsense-viewer" and "rs-multicam". Both the devices are streaming simultaneously when viewed using **rs-multicam** app. In case of **realsense-viewer** it connects to just one device among the two. 

The cameras are on separate USB3 ports, not connected through a USB hub.

The following are the logs to the above mentioned commands.

**lsusb**
```
Bus 002 Device 003: ID 8086:0ad3 Intel Corp. 
Bus 002 Device 002: ID 8086:0ad3 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 004: ID 110a:1110 Moxa Technologies Co., Ltd. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**usb-devices**
```
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh=12
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.10
S:  Manufacturer=Linux 4.10.0-35-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:00:14.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=01 Prnt=01 Port=03 Cnt=01 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=110a ProdID=1110 Rev=02.00
S:  Manufacturer=Moxa Technologies Co., Ltd.
S:  Product=UPort 1110
C:  #Ifs= 1 Cfg#= 2 Atr=80 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=ti_usb_3410_5052

T:  Bus=01 Lev=01 Prnt=01 Port=07 Cnt=02 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=0a2b Rev=00.10
C:  #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
I:  If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 6
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev=04.10
S:  Manufacturer=Linux 4.10.0-35-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:00:14.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=8086 ProdID=0ad3 Rev=50.9b
S:  Manufacturer=Intel(R) RealSense(TM) 415
S:  Product=Intel(R) RealSense(TM) 415
S:  SerialNumber=806413021184
C:  #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=440mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 4 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo

T:  Bus=02 Lev=01 Prnt=01 Port=01 Cnt=02 Dev#=  3 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=8086 ProdID=0ad3 Rev=50.9b
S:  Manufacturer=Intel(R) RealSense(TM) 415
S:  Product=Intel(R) RealSense(TM) 415
S:  SerialNumber=806413020449
C:  #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=440mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 4 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
```

Just to reiterate, I am facing the problem with the ROS launch file not being able to stream data simultaneously. Also we have made the necessary changes in code due to the hardware reset problem as mentioned in the issue [#314](https://github.com/intel-ros/realsense/issues/314). The changes are made as per the suggestions by @icarpis in the thread. 

Looking forward to your support.

Thanks.

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
Hi @ankurbaruah90,

any update? Still need further support for this issue?

ankurbaruah90, 2018-06-29:
Hi @RealSense-Customer-Engineering ,
The current configuration on my setup is:
ros wrapper v2.0.3 + ros kinetic + librealsense v2.12.0 + kernel 4.10.0-35-generic
And no rs_multiple_devices.launch still cannot launch two D415 devices together. However, a delay added to the system while launching them both does the job. I will see if updating the kernel solves the problem. Will let you know if any problem turns up in the future. 
Thanks for your support.

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
Hi @ankurbaruah90,

any finding after kernel updating?

ankurbaruah90, 2018-07-16:
@RealSense-Customer-Engineering,

I have not updated the kernel yet. You can close the issue for now. The workaround seems to be working fine. Thanks for you help.

closed 2018-07-16 by RealSense-Customer-Engineering

RachitB11, 2019-01-14:
@ankurbaruah90 Hey, I'm facing a similar issue. How do you add a delay to the system?

ankurbaruah90, 2019-01-21:
> @ankurbaruah90 Hey, I'm facing a similar issue. How do you add a delay to the system?

@RachitB11 Use a bash or python script to launch each node individually, adding a delay between them. Here is the snippet of my bash script.

```
#!/bin/bash

roslaunch realsense2_camera rs_camera_1.launch &
sleep 10
roslaunch realsense2_camera rs_camera_2.launch &
 ```

RachitB11, 2019-01-21:
@ankurbaruah90 thanks for the reply. 
I added the sleep in the realsense_node_factory.cpp after I call the reset. That seems to work for me.